chipweinberger/flutter_blue_plus #2: MacOS support

ekuleshov, 2022-01-31:
The original `flutter_blue` plugin had MacOS support (somewhat broken after moving to null safety). Please add this functionality to the `flutter_blue_plus`

boskokg, 2022-01-31:
@ekuleshov I saw that there were several issues reported that MacOS doesn't work, so I didn't just copy that part of the code. I will do add it in the next version and validated that it actually works.
I saw that other plugins done separation of platforms in the separate subplugins, so maybe we should do the same thing here...

ekuleshov, 2022-01-31:
> I saw that other plugins done separation of platforms in the separate subplugins, so maybe we should do the same thing here...

Moving to [federated plugins](https://docs.flutter.dev/development/packages-and-plugins/developing-packages#federated-plugins) is a bigger task, imho. It is a good idea for the long run maintenance and to support other platforms, but perhaps do it as a separate refactoring task after bringing the existing macos code in.

FXschwartz, 2022-02-04:
Looks like possible some amount of effort went into moving to a [federated plugin setup in the old flutter_blue](https://github.com/pauldemarco/flutter_blue/issues/778), I wonder if we could pick up where they left off.

ekuleshov, 2022-02-05:
> Looks like possible some amount of effort went into moving to a [federated plugin setup in the old flutter_blue](https://github.com/pauldemarco/flutter_blue/issues/778), I wonder if we could pick up where they left off.

What I see there is that android, ios and macos implementations are still under the same module, and they just extracted a platform interface and did a separate module/plugin for linux.

All in all, it still better to bring back the macos first and then refactor to federated or pull in that linux branch, if developers are still around (it been 11 months since last change out there).

chipweinberger, 2023-07-18:
macOS support has landed. 

use 1.7.4.

closed 2023-07-18 by chipweinberger