chipweinberger/flutter_blue_plus #5: Caching GATT on IOS...

thormj, 2022-02-03:
Moved from FlutterBlue, still having the same problem...
2A00 -- Device Name isn't showing up in the GATT.  It is in there in reality and is visible on Android.
I've tried turning Bluetooth on/off and rebooting the iPhone...

<I'm going to update this with Android run.... gimme a few minutes>

IOS Log:
2022-02-03 16:29:26.178508-0600 Runner[1020:128813] flutter: 2022-02-03 16:29:26.176052 Connecting to 8FF8E2CD-E1FA-A880-0D3B-622622F2F7E1...
2022-02-03 16:29:26.179884-0600 Runner[1020:128813] flutter: WannaConnect
2022-02-03 16:29:26.179987-0600 Runner[1020:128813] flutter: GonnaConnect
2022-02-03 16:29:26.838047-0600 Runner[1020:128483] didConnectPeripheral
2022-02-03 16:29:26.849271-0600 Runner[1020:128813] flutter: then
2022-02-03 16:29:26.850926-0600 Runner[1020:128813] flutter: return
2022-02-03 16:29:26.851550-0600 Runner[1020:128813] flutter: 2022-02-03 16:29:26.851442 ConnectDone
2022-02-03 16:29:27.531254-0600 Runner[1020:128483] didDiscoverServices
2022-02-03 16:29:27.531635-0600 Runner[1020:128483] Found service: E4CF7947-AD98-425B-8F0C-571F873A88FC
2022-02-03 16:29:27.531930-0600 Runner[1020:128483] Found service: 180A
2022-02-03 16:29:27.532084-0600 Runner[1020:128483] Found service: 1815
2022-02-03 16:29:27.532205-0600 Runner[1020:128483] Found service: 3A9057FB-6F8A-4D7E-AD45-62B1E4B2F4F3
2022-02-03 16:29:27.532319-0600 Runner[1020:128483] Found service: 3A9057FB-6F8A-4F7E-AE45-62BCE4B2F4F1
2022-02-03 16:29:27.532441-0600 Runner[1020:128483] Found service: 1805
2022-02-03 16:29:27.532580-0600 Runner[1020:128483] Found service: 181A
2022-02-03 16:29:27.532701-0600 Runner[1020:128483] Found service: 1828
2022-02-03 16:29:27.532822-0600 Runner[1020:128483] Found service: 1D14D6EE-FD63-4FA1-BFA4-8F47B42119F0
2022-02-03 16:29:27.708441-0600 Runner[1020:128483] didDiscoverCharacteristicsForService
2022-02-03 16:29:27.889213-0600 Runner[1020:128483] didDiscoverCharacteristicsForService
2022-02-03 16:29:27.981268-0600 Runner[1020:128483] didDiscoverCharacteristicsForService
2022-02-03 16:29:28.039130-0600 Runner[1020:128483] didDiscoverCharacteristicsForService
2022-02-03 16:29:28.340595-0600 Runner[1020:128483] didDiscoverCharacteristicsForService
2022-02-03 16:29:28.398334-0600 Runner[1020:128483] didDiscoverCharacteristicsForService
2022-02-03 16:29:28.518781-0600 Runner[1020:128483] didDiscoverCharacteristicsForService
2022-02-03 16:29:28.608596-0600 Runner[1020:128483] didDiscoverCharacteristicsForService
2022-02-03 16:29:28.728428-0600 Runner[1020:128483] didDiscoverCharacteristicsForService
2022-02-03 16:29:28.787918-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.788080-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.847465-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.847588-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.847634-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.847672-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.847704-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.847738-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.847771-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.847975-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.848096-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.848161-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.848227-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.848286-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.907979-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:28.967715-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.087787-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.148049-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.208359-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.267973-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.328643-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.417654-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.478226-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.538169-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.598341-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.658771-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.658975-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.659059-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.659138-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.659214-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.659289-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.718429-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.718652-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.718736-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.718815-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.718891-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
2022-02-03 16:29:29.807972-0600 Runner[1020:128483] didDiscoverDescriptorsForCharacteristic
[... 153 lines cut ...]
2022-02-03 16:29:30.569869-0600 Runner[1020:128813] flutter:       7d311226-c6fd-4a75-8abb-41404c08b67e
2022-02-03 16:29:30.569919-0600 Runner[1020:128813] flutter:       df68adc1-3c71-4e2b-a468-90dc765b07a7
2022-02-03 16:29:30.570168-0600 Runner[1020:128813] flutter:       f35be144-5659-4cf1-86e2-fe7d1c82de20
2022-02-03 16:29:30.570237-0600 Runner[1020:128813] flutter:       d2ac2d86-bd61-44cc-924b-2ae2ed221680
2022-02-03 16:29:30.570368-0600 Runner[1020:128813] flutter:       2747a34e-e277-4510-98b0-91c176d57a61
2022-02-03 16:29:30.570440-0600 Runner[1020:128813] flutter:       57df3786-7b57-472b-bd5a-a4bc758952ab
2022-02-03 16:29:30.570485-0600 Runner[1020:128813] flutter:       c72e32ee-e309-4f32-b83f-8ef7e7b6f4e8
2022-02-03 16:29:30.570674-0600 Runner[1020:128813] flutter:       d709cb47-3b4c-4f66-9588-40cd0a95d14d
2022-02-03 16:29:30.570743-0600 Runner[1020:128813] flutter:       56eb7023-0b8b-450d-89bd-3b2881b5d327
2022-02-03 16:29:30.570868-0600 Runner[1020:128813] flutter:       5b6487e1-8bae-4dc1-b86a-867806a9de22
2022-02-03 16:29:30.570932-0600 Runner[1020:128813] flutter:   00001805-0000-1000-8000-00805f9b34fb
2022-02-03 16:29:30.571072-0600 Runner[1020:128813] flutter:       00002a2b-0000-1000-8000-00805f9b34fb
2022-02-03 16:29:30.571139-0600 Runner[1020:128813] flutter:   0000181a-0000-1000-8000-00805f9b34fb
2022-02-03 16:29:30.571186-0600 Runner[1020:128813] flutter:       5e1ab631-9c07-4715-9d16-c235294cd3e6
2022-02-03 16:29:30.571222-0600 Runner[1020:128813] flutter:       00005501-0000-1000-8000-00805f9b34fb
2022-02-03 16:29:30.571256-0600 Runner[1020:128813] flutter:       00005502-0000-1000-8000-00805f9b34fb
2022-02-03 16:29:30.571292-0600 Runner[1020:128813] flutter:       00005503-0000-1000-8000-00805f9b34fb
2022-02-03 16:29:30.571325-0600 Runner[1020:128813] flutter:       00008500-0000-1000-8000-00805f9b34fb
2022-02-03 16:29:30.571362-0600 Runner[1020:128813] flutter:   00001828-0000-1000-8000-00805f9b34fb
2022-02-03 16:29:30.571394-0600 Runner[1020:128813] flutter:       00002add-0000-1000-8000-00805f9b34fb
2022-02-03 16:29:30.571426-0600 Runner[1020:128813] flutter:       00002ade-0000-1000-8000-00805f9b34fb
2022-02-03 16:29:30.571471-0600 Runner[1020:128813] flutter:   1d14d6ee-fd63-4fa1-bfa4-8f47b42119f0
2022-02-03 16:29:30.571507-0600 Runner[1020:128813] flutter:       f7bf3564-fb6d-4e53-88a4-5e37e0326063
2022-02-03 16:29:30.571550-0600 Runner[1020:128813] flutter: E:--endofline--
2022-02-03 16:29:30.571611-0600 Runner[1020:128813] flutter: 2022-02-03 16:29:30.571572 !  BC NotFound:Name[2A00]

thormj, 2022-02-04:
Ahh... it looks like IOS hides the entire 1800 service from me (Device Name, Appearance).  Looks like I need to duplicate the characteristic to a "more visible place..."

closed 2022-02-04 by thormj

boskokg, 2022-02-04:
Some asked [here](https://lists.apple.com/archives/bluetooth-dev/2016/Feb/msg00018.html) for the same issue, I think...

thormj, 2022-02-04:
And that worked... so... there are things IOS keeps secret from people... the more you know....