chipweinberger/flutter_blue_plus #13: FlutterBluePlus.state behaves strange.

maciej-rexton, 2022-02-24:
There were similar issues in the upstream repository already:

- [Stream<BluetoothState> get state does not return a broadcast stream](https://github.com/pauldemarco/flutter_blue/issues/767)
- [FlutterBlue.state cancels all subscriptions if only one listener cancels](https://github.com/pauldemarco/flutter_blue/issues/608)

And it's still an issue. In my case a class that is provided with `provider` package at the root of the widget tree is trying to keep connection with any device that can act as a gateway to a mesh network between IoT devices. It's doing this in the background and listens to the `state` field to do it only when Bluetooth is turned on. 

In the same time widgets that require gateway connection, so user can make any interaction with the network, are listening `state` to indicate that gateway is not available, because Bluetooth is turned off (other reasons may be that it's just not around or location services are disabled). Of course it's abstracted to a single class like `BluetoothNotAvailableDialog`, but this still leaves a single instance of that service class and multiple instances of the dialog class and they cannot listen to `state` field at the same time, because only one of them receives events and the rest do not.

There already exists a [pull request which fixes the issue](https://github.com/pauldemarco/flutter_blue/pull/609). It's pretty small change so should be easy to review and merge.

SeekayOng, 2022-03-17:
Kindly solve this issue please

MrCsabaToth, 2022-06-27:
Please merge that referenced PR https://github.com/pauldemarco/flutter_blue/pull/609

boskokg, 2022-07-23:
Thanks. commited

closed 2022-07-23 by boskokg

MrCsabaToth, 2022-07-23:
Yes! My custom fork is now closer to the upstream!