chipweinberger/flutter_blue_plus #14: Package does not have a verified uploader according to Pub

navaronbracke, 2022-02-27:
Apparently, the package has an `unverified uploader` according to Pub.

Can we fix that?

Related docs https://dart.dev/tools/pub/verified-publishers

boskokg, 2022-04-10:
Hi @navaronbracke. Is this needed to anyone? The purpose of flutter_blue_plus is to be maintained by anyone, not only by me... You can see from the changelog that most of the fixes are done by someone else.

chipweinberger, 2023-07-26:
@boskokg , could you move this package to my verified uploader? I'd love for this FlutterBluePlus to be more legitimate.

https://pub.dev/publishers/jamcorder.com/packages

[image]

chipweinberger, 2023-07-26:
Also, it would be useful to add other github contributers, and to be able to make an issue template.

chipweinberger, 2023-10-12:
I have added  flutter_blue_plus to my publisher: `jamcorder.com`

https://pub.dev/publishers/jamcorder.com/packages

closed 2023-10-12 by chipweinberger