chipweinberger/flutter_blue_plus #21: Android 12 not detecting beacons and some of the BLE devices.

watsmyname, 2022-03-06:
This plugin works, but not as intended in Android 12.  

In Android 12, the scan results in devices, but it is not discovering the beacons and some BLE devices. However same code in lower android versions discovers them. What I am doing wrong? Or is it an issue with plugin?

I have following in AndroidManifest.xml.

```

    <uses-permission android:name="android.permission.BLUETOOTH"
                     android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
                     android:maxSdkVersion="30" />

   
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />

    <!-- Needed only if your app communicates with already-paired Bluetooth
         devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.FOREGROUND_SERVICE" />
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
```

And in my code 

I have requested permissions like this before scanning.

```
final permissionBT = await Permission.bluetooth.request().isGranted;
    final permissionBTScan = await Permission.bluetoothScan.request().isGranted;
    final permissionBTConnect =
        await Permission.bluetoothConnect.request().isGranted;
    final permissionBTAd =
        await Permission.bluetoothAdvertise.request().isGranted;
    final permissionLoc = await Permission.location.request().isGranted;
    final permissionLocALways =
        await Permission.locationAlways.request().isGranted;
    final permissionLocInUse = await Permission.locationWhenInUse.request().isGranted;
```

gregk104, 2022-03-06:
If your manifest says "neverForLocation", but then you ask for it in the code, what is the result?  What happens if you remove the "neverForLocation" permission flag?

Since beacons can be/used for location, this might be your problem.

watsmyname, 2022-03-06:
> android:usesPermissionFlags="neverForLocation"

Tried removing `android:usesPermissionFlags="neverForLocation"` as well, but doesn't work, no errors either.

gregk104, 2022-03-06:
Although this may be true of iOS as well, one very annoying behavior I found with Android 12, is that it remembers permission sets.  I think this aims to make things easier for the user.  Try uninstalling the app from the phone, then retrying the test.

What was the value of permissionLoc?

I'm not sure any of this is going to solve your problem, but it mirrors the actions I went through to fix our app for Android 12.  We don't use beacons; we're a device manufacturer, and 12 was preventing our users from connecting to our stuff. 

One other thing, simply switching to FlutterBluePlus did not solve the problem.  Ultimately, we had to handle the permissions ourselves.  We still moved to FlutterBluePlus because we want a package that has an active author and community.

watsmyname, 2022-03-06:
> 

The value of `permissionLoc` is true. I tried flutter cleaning, removing the app from device and rebuilding it in DEV as well as release build. But with no luck.

watsmyname, 2022-03-07:
I fixed the issue

I updated `AndroidManifest.xml` file of a  `flutter_blue_plus` plugin. I removed all the permissions and pasted the following

```
<uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />

  <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
  <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
```

The real culprit was

`android:usesPermissionFlags="neverForLocation"`

this line caused the issue of not detecting sensors and beacons.

sophisticode, 2022-03-07:
The proper fix for this issue is to merge the permissions for the Manifest file correctly. During the build process, the information from the Manifest files in the different packages are merged into one. The final step of that merging process can be influenced in the top level (app specific) manifest file.
The merging rules are described in this document: https://developer.android.com/studio/build/manage-manifests

The best way to remove the flag "neverForLocation" is to replace the previously defined permission (that has this flag) completely by the following statement in the app specific Manifest file (android/app/src/main/AndroidManifest.xml):

<uses-permission android:name="android.permission.BLUETOOTH_SCAN" tools:node="replace" />

With this change, the merged Manifest file will not have the flag "neverForLocation" anymore. You can check the merged Manifest file in the build/app/intermediates/merged_manifest directory.

watsmyname, 2022-03-10:
Yes that will work, but not to miss will need to add this  in Manifest tag of app's AndroidManifest.xml

`xmlns:tools="http://schemas.android.com/tools"`

closed 2022-03-10 by watsmyname