chipweinberger/flutter_blue_plus #23: Getting rssi continously for scanResults?

watsmyname, 2022-03-10:
Actually this is more like a question than an issue.

What i need to do is scan for bluetooth devices, And I periodically need to  update the list with new device, and for existing device, updating their corresponding rssi in the list. For this I use Timer to execute findBluetoothPeripherial(), every 40 seconds. 
My Question, is there a better way of achieving this? 

```
FlutterBluePlus flutterBlue = FlutterBluePlus.instance;
List scanResults = <ScanResult>[];
Timer? timer;

@override
void initState(){
	super.initState();
	findBluetoothPeripherial();

	timer = Timer.periodic(Duration(seconds: 40), (Timer t) => findBluetoothPeripherial());
}

findBluetoothPeripherial(){
    //check and handle permissions
	flutterBlue.startScan(timeout: Duration(seconds: 20));

	var subscription = flutterBlue.scanResults.listen((results) {
    
	    for (ScanResult r in results) {
	        print('${r.device.name} found! rssi: ${r.rssi}');
	        int index = scanResults.indexWhere((item) => item.device.id == r.device.id);

	        if(index > 0){
	        	setState((){
		        	scanResults[index].rssi = r.rssi;
	        	});
	        }else {
	        	setState(() {
	        		scanResults.add(r);
	        	});
	        }
	    }
	});

	// Stop scanning
	flutterBlue.stopScan();
}
```

Another question

What is the difference between -

```
flutterBlue.scanResults.listen((results){
	//code here
});
```

And

```
flutterBlue.scan(timeout: Duration(seconds: 20))
        .distinct()
        .asBroadcastStream()
        .listen((results){
        		//code here
        });
```

Mr-xzq, 2022-04-20:
I also had this problem that the scan couldn't keep up with the latest devices around me.
My temporary solution is the same as yours, which is to restart the scan periodically.

SaikCaskey, 2022-08-17:
you could try `allowDuplicates`, it's not documented, but allows duplicate scan results to be reported (i.e., > 1 for same device)

chipweinberger, 2023-07-20:
yes that sounds good to me^

Please try 1.7.6+. reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger