chipweinberger/flutter_blue_plus #27: How to read more than 20 bytes data (MTU negotation is not available)

flerik12, 2022-03-21:
Hello,

I have a problem to read the proper amount of data sent by BLE peripheral. The peripheral cannot send bigger packages than 20 bytes (MTU is not negotiable). It cuts every message into 20 bytes packets. Unfortunately, I am not able to catch the first few packets, just the last one. While i subscribe to a characteristic and force it to send data, I can't find a way to catch the single packets. I met similar problem on pure Android app, but there I was able to save each packet from a characteristic and then reuse it. Here, it seems like they disappear and only the last one is used. Any ideas in this case?
I will be very grateful for any insights.

closed 2022-03-22 by flerik12

flerik12, 2022-06-01:
Hello @aaronchen96 
You have to subscribe to the characteristic instead of performing just a casual read. When the peripheral will start transferring data, your subscribed characteristic will receive each packet, instead of just the last one which is saved in it's memory.

aaronchen96, 2022-06-01:
@flerik12  thank you for ur reply and this really help. Also i realised that i need to set config the device MTU size.