chipweinberger/flutter_blue_plus #28: Can't instantiate FlutterBluePlus in an Isolate function

calebadavis, 2022-03-23:
Calling 'FlutterBluePlus.instance' asserts with: Cannot set the method call handler before the binary messenger has been initialized. This happens when you call setMethodCallHandler() before the WidgetsFlutterBinding has been initialized. You can fix this by either calling WidgetsFlutterBinding.ensureInitialized() before this or by passing a custom BinaryMessenger instance to MethodChannel().

calebadavis, 2022-03-23:
I noticed an old issue on the old 'flutter_blue' project which has sat untouched for two years. Is there any chance this could be made to work here?

boskokg, 2022-05-08:
@calebadavis please create merge request regarding this issue.

closed 2022-05-08 by boskokg