chipweinberger/flutter_blue_plus #34: Ios: Every time you open the APP, the first scan always fails, and the second scan is normal

jahnli, 2022-03-30:
Ios: Every time you open the APP, the first scan always fails, and the second scan is normal  

This problem does not exist with Flutter Blue

jahnli, 2022-03-30:
@boskokg   Ios: Every time you open the APP, the first scan always fails, and the second scan is normal

jahnli, 2022-03-30:
first scan  :
[image]

jahnli, 2022-03-30:
two scan 
[image]

JaysonJia, 2022-04-04:
I had the same issue: when first startscan:
'[CoreBluetooth] API MISUSE: <CBCentralManager: 0x282854480> can only accept this command while in the powered on state'

MATTYGILO, 2022-04-05:
Having the same problem, any updates?

boskokg, 2022-04-10:
Can you please debug the problem and propose the merge request of the solution?

eriksoderstrom, 2022-04-20:
try {
      await for(final state in _flutterBlue.state.timeout(const Duration(seconds: 5))) {
        if(state == bt.BluetoothState.on) {
          _logger.info('BT Adapter turned ON');
          break;
        }
      }
    } catch(e) {
      _logger.info('Timeout when waiting for BT adapter ON');
    }
    return _flutterBlue.startScan(timeout: timeout, withServices: [typeToScan]);

I've used this to wait until BT adapter switches ON for iOS.

Hitbee-dev, 2022-05-19:
``` dart
@override
initState() {
  super.initState();
Future.delayed(const Duration(milliseconds: 100), () {
      scan();
    });
}
```

It was resolved by delaying `100ms` when entering the Bluetooth settings page and scanning.

closed 2022-07-23 by boskokg