chipweinberger/flutter_blue_plus #36: PlatformException(connect, Peripheral not found, null)

WeiKuang, 2022-03-30:
My flow was:

1、Init & scan etc
2、Connect device
3、Discover services & Characteristics etc
4、Do stuff
5、Disconnect all BT devices (triggered by app entering background 'paused' state)
6、Resume app
7、Init and scan etc
=> note: after finding the correct device (and before connecting or calling discover) it still had the list of Services & Characteristics that were found previously... this is a red flag (these should be empty).
Connect device ==> PlatformException(connect, Peripheral not found, null)
My workaround is to add this BlowItUp method to the FluttereBlue class and call it before getting FlutterBlue.instance whenever I connect to BT. This ensures each time I try to reconnect I get a freshly instantiated FlutterBlue.

[https://github.com/pauldemarco/flutter_blue/issues/497](url)

toxamontsg, 2022-04-03:
Similar problem. I plug in the device and everything works fine. I added automatic reconnect to ios [3f5b937](https://github.com/boskokg/flutter_blue/pull/10/commits/3f5b9378077b71f3f2ceda8c743b634a1892702e) and when reconnecting, this error pops up.

WeiKuang, 2022-04-05:
I found the problem. The reason is because when Bluetooth starts scanning, if the scan result is monitored immediately, an exception will be thrown in the ios file. This problem does not appear at the beginning of the scan, but only when the bluetooth is disconnected and re-scanned to connect.

closed 2022-05-27 by boskokg

mevluttosun, 2022-09-13:
I've still same issue. Maybe I am using proto to connect device. In android I have no issue but in my test device IOS 12.5.5, ( I hadn't try with another device) always giving same PlatformException(connect, Peripheral not found, null, null)

Here is my proto extention

extension BluetoothDeviceExtension on BluetoothDeviceModel { BluetoothDevice toBluetoothDevice() { proto.BluetoothDevice p = proto.BluetoothDevice.create(); 
p.name = name; 
p.type = proto.BluetoothDevice_Type.LE;
p.remoteId = id; 
return BluetoothDevice.fromProto(p);
 } }

Here is my connect method

Future<void> connect(bool autoConnect) async { 
log.fine('connecting'); 
emit(const BluetoothModelDeviceState.connecting());
final bluetoothDevice = device.toBluetoothDevice();
await bluetoothDevice .connect( autoConnect: autoConnect, ) 
.onError((error, stackTrace) => log.warning('Connection Error ${error.toString()}')); }

WeiKuang, 2023-04-12:
iOS is not working from proto

ryanhz, 2023-05-24:
> I found the problem. The reason is because when Bluetooth starts scanning, if the scan result is monitored immediately, an exception will be thrown in the ios file. This problem does not appear at the beginning of the scan, but only when the bluetooth is disconnected and re-scanned to connect.

This is not the reason and cause of the problem, this issue should not be closed

keskink, 2023-06-24:
Hello friends. I fixed the "peripheral not found" issue.
Now I can connect/reconnect to the device without scanning, also for iOS.
this plugin can already connect without scanning on Android.

I can share my solution with plugin owners.