chipweinberger/flutter_blue_plus #37: Maybe a question, what's difference between flutter_blue and flutter_blue_plus?

JaysonJia, 2022-04-01:
For now, my project is using flutter_blue, and i see there are some improvements on flutter_blue_plus, so i want to use flutter_blue_plus instead of flutter_blue,  do you think it is possible and with less code changes?

boskokg, 2022-04-01:
@JaysonJia flutter_blue is not maintained for a long time. We had major issues with M1 chip and Android 12 support and we created flutter_blue_plus but also added other improvements.  Mostly there should be no major changes when migrating to flutter_blue_plus.

closed 2022-04-01 by boskokg

JaysonJia, 2022-04-01:
> @JaysonJia flutter_blue is not maintained for a long time. We had major issues with M1 chip and Android 12 support and we created flutter_blue_plus but also added other improvements. Mostly there should be no major changes when migrating to flutter_blue_plus.

Thanks!