chipweinberger/flutter_blue_plus #38: iOS can't read descriptors value, works perfect for Android.

abcdggs, 2022-04-01:
As the title suggests I am trying to read the value of descriptors for characteristics. This works fine on android but doesn't seem to work for iOS. Has anyone experienced or overcome this issue?

janleskovec, 2022-05-17:
It might be related to this issue: https://github.com/pauldemarco/flutter_blue/issues/760 .
If the device is already connected or paired, you can't seem to be able to discover services.

unitedsoftworks, 2022-05-22:
currently facing this issue on iOS and not android.

AymenXD, 2023-02-24:
same goes for me, this is an example:

Android (works fine):
[image]

iOS:
[image]

jojoX94, 2023-04-17:
Same issue on iOS.

chipweinberger, 2023-07-20:
Please try 1.7.6+. reopen a new issue if you still have problems.

And feel free to open a PR.

closed 2023-07-20 by chipweinberger