chipweinberger/flutter_blue_plus #45: cannot get indicate value listener

aaronchen96, 2022-04-14:
for my module, i first need to write to first characteristic, follow by set second characteristic indicate to true, and listen to the indicate return value. But i failed to get the return value.

aaronchen96, 2022-04-14:
getConnectedDevice() async {
    flutterBlue.connectedDevices.then((devices) async {
      List<BluetoothCharacteristic> uids = [];
      List<String> requiredid = ['00001212', '00002a35'];
      for (BluetoothDevice device in devices) {
        await device.discoverServices().then((services) async {
          for (BluetoothService service in services) {
            service.characteristics.forEach((element) {
              if (requiredid
                  .contains(element.uuid.toString().substring(0, 8))) {
                uids.add(element);
              }
            });
          }
        });
        print(uids.map((e) => e.uuid));
        uids
            .firstWhere(
                (element) => element.uuid.toString().contains('00001212'))
            .write([0xA504120016], withoutResponse: true).then((value) async {
          var c = uids.firstWhere(
              (element) => element.uuid.toString().contains('00002a35'));
          await c.setNotifyValue(true);
          print(c);
          Future.delayed(const Duration(milliseconds: 2000));
          await c.value.listen((l) {
            print('result' + l.toList().toString());
            
          });
        });
      }
    });
  }

aaronchen96, 2022-04-14:
[image]

aaronchen96, 2022-04-14:
[image]

aaronchen96, 2022-04-18:
> getConnectedDevice() async { flutterBlue.connectedDevices.then((devices) async { List uids = []; List requiredid = ['00001212', '00002a35']; for (BluetoothDevice device in devices) { await device.discoverServices().then((services) async { for (BluetoothService service in services) { service.characteristics.forEach((element) { if (requiredid .contains(element.uuid.toString().substring(0, 8))) { uids.add(element); } }); } }); print(uids.map((e) => e.uuid)); uids .firstWhere( (element) => element.uuid.toString().contains('00001212')) .write([0xA504120016], withoutResponse: true).then((value) async { var c = uids.firstWhere( (element) => element.uuid.toString().contains('00002a35')); await c.setNotifyValue(true); print(c); Future.delayed(const Duration(milliseconds: 2000)); await c.value.listen((l) { print('result' + l.toList().toString());
> 
> ```
>       });
>     });
>   }
> });
> ```
> 
> }

closed 2022-04-20 by aaronchen96

massimilianochiodi, 2022-09-14:
I have some issues on IOS 

`2022-09-14 14:42:57.398747+0200 Power UP![897:87822] didDiscoverServices
2022-09-14 14:42:57.398852+0200 Power UP![897:87822] Found service: 180A
2022-09-14 14:42:57.398942+0200 Power UP![897:87822] Found service: 331A36F5-2459-45EA-9D95-6142F0C4B307
2022-09-14 14:42:57.399003+0200 Power UP![897:87822] Found service: 169B52A0-B7FD-40DA-998C-DD9238327E55
2022-09-14 14:42:57.399488+0200 Power UP![897:87822] didDiscoverCharacteristicsForService
2022-09-14 14:42:57.399712+0200 Power UP![897:87822] didDiscoverCharacteristicsForService
2022-09-14 14:42:57.399865+0200 Power UP![897:87822] didDiscoverCharacteristicsForService
2022-09-14 14:42:57.399971+0200 Power UP![897:87822] didDiscoverDescriptorsForCharacteristic
2022-09-14 14:42:57.399996+0200 Power UP![897:87822] didDiscoverDescriptorsForCharacteristic
2022-09-14 14:42:57.400015+0200 Power UP![897:87822] didDiscoverDescriptorsForCharacteristic
2022-09-14 14:42:57.400201+0200 Power UP![897:87822] didDiscoverDescriptorsForCharacteristic
2022-09-14 14:42:57.400391+0200 Power UP![897:87822] didDiscoverDescriptorsForCharacteristic
2022-09-14 14:42:57.400527+0200 Power UP![897:87822] didDiscoverDescriptorsForCharacteristic
2022-09-14 14:42:57.400550+0200 Power UP![897:87822] didDiscoverDescriptorsForCharacteristic
2022-09-14 14:42:57.400602+0200 Power UP![897:87822] didDiscoverDescriptorsForCharacteristic
2022-09-14 14:42:57.400676+0200 Power UP![897:87822] didDiscoverDescriptorsForCharacteristic
2022-09-14 14:42:57.400739+0200 Power UP![897:87822] peripheral uuid:A6968D4E-5AB2-22A1-51CF-3DAAC313BE01
2022-09-14 14:42:57.400775+0200 Power UP![897:87822] service uuid:0000180a-0000-1000-8000-00805f9b34fb
2022-09-14 14:42:57.400874+0200 Power UP![897:87822] uuid: 00002a29-0000-1000-8000-00805f9b34fb value: (null)
2022-09-14 14:42:57.400912+0200 Power UP![897:87822] uuid: 00002a24-0000-1000-8000-00805f9b34fb value: (null)
2022-09-14 14:42:57.400942+0200 Power UP![897:87822] uuid: 00002a26-0000-1000-8000-00805f9b34fb value: (null)
2022-09-14 14:42:57.400968+0200 Power UP![897:87822] peripheral uuid:A6968D4E-5AB2-22A1-51CF-3DAAC313BE01
2022-09-14 14:42:57.400985+0200 Power UP![897:87822] service uuid:331a36f5-2459-45ea-9d95-6142f0c4b307
2022-09-14 14:42:57.401029+0200 Power UP![897:87822] uuid: a9da6040-0823-4995-94ec-9ce41ca28833 value: {length = 3, bytes = 0x00ff07}
2022-09-14 14:42:57.401079+0200 Power UP![897:87822] uuid: a73e9a10-628f-4494-a099-12efaf72258f value: (null)
2022-09-14 14:42:57.401119+0200 Power UP![897:87822] uuid: 75a9f022-af03-4e41-b4bc-9de90a47d50b value: {length = 1, bytes = 0x01}
2022-09-14 14:42:57.401176+0200 Power UP![897:87822] peripheral uuid:A6968D4E-5AB2-22A1-51CF-3DAAC313BE01
2022-09-14 14:42:57.401235+0200 Power UP![897:87822] service uuid:169b52a0-b7fd-40da-998c-dd9238327e55
2022-09-14 14:42:57.401290+0200 Power UP![897:87822] uuid: 902ee692-6ef9-48a8-a430-5212eeb3e5a2 value: (null)
2022-09-14 14:42:57.401336+0200 Power UP![897:87822] uuid: 503a5d70-b443-466e-9aeb-c342802b184e value: (null)
2022-09-14 14:42:57.401384+0200 Power UP![897:87822] uuid: 12e868e7-c926-4906-96c8-a7ee81d4b1b3 value: (null)
2022-09-14 14:42:57.862409+0200 Power UP![897:88380] flutter: Servizio trovato 0000180a-0000-1000-8000-00805f9b34fb
2022-09-14 14:42:57.862614+0200 Power UP![897:88380] flutter: Servizio trovato 331a36f5-2459-45ea-9d95-6142f0c4b307
2022-09-14 14:42:57.862743+0200 Power UP![897:88380] flutter: Servizio >>>> scansione caratteristiche 331a36f5-2459-45ea-9d95-6142f0c4b307
2022-09-14 14:42:57.863495+0200 Power UP![897:88380] flutter: Caratteristica Trovata a9da6040-0823-4995-94ec-9ce41ca28833
2022-09-14 14:42:57.863638+0200 Power UP![897:88380] flutter: Caratteristica >>>> BLUETOOTH_WALLBOX_RX
2022-09-14 14:42:57.863780+0200 Power UP![897:88380] flutter: Caratteristica Trovata a73e9a10-628f-4494-a099-12efaf72258f
2022-09-14 14:42:57.863882+0200 Power UP![897:88380] flutter: Caratteristica >>>> BLUETOOTH_WALLBOX_TX
2022-09-14 14:42:57.863997+0200 Power UP![897:88380] flutter: Caratteristica Trovata 75a9f022-af03-4e41-b4bc-9de90a47d50b
2022-09-14 14:42:57.864070+0200 Power UP![897:88380] flutter: Caratteristica >>>> BLUETOOTH_WALLBOX_ST
2022-09-14 14:42:57.866355+0200 Power UP![897:88380] flutter: Imposto canale notifica
2022-09-14 14:42:57.867477+0200 Power UP![897:88380] flutter: Servizio trovato 169b52a0-b7fd-40da-998c-dd9238327e55
2022-09-14 14:42:57.867635+0200 Power UP![897:88380] flutter: Indicate rxdata: ricevuto <<
2022-09-14 14:42:57.893614+0200 Power UP![897:87822] didUpdateNotificationStateForCharacteristic
2022-09-14 14:42:57.924313+0200 Power UP![897:87822] uuid: a73e9a10-628f-4494-a099-12efaf72258f value: (null)
2022-09-14 14:42:57.928453+0200 Power UP![897:88380] flutter: Notifca attivata
2022-09-14 14:42:57.933129+0200 Power UP![897:88380] flutter: Provo a richiedere l'auth come service BLE,SERV,2222
2022-09-14 14:42:57.933552+0200 Power UP![897:88380] flutter: Scrivo $BLE,SERV,2222
2022-09-14 14:42:57.935633+0200 Power UP![897:88380] flutter: valore ricevuto []
2022-09-14 14:42:57.935685+0200 Power UP![897:88380] flutter: Ricevo da txdata: ricevuto <<  >>
2022-09-14 14:42:57.969664+0200 Power UP![897:87822] didWriteValueForCharacteristic
2022-09-14 14:42:57.971681+0200 Power UP![897:88380] flutter: Ho scritto = $BLE,SERV,2222
[... cut ...]
2022-09-14 14:42:57.973186+0200 Power UP![897:87822] uuid: a73e9a10-628f-4494-a099-12efaf72258f value: (null)
2022-09-14 14:42:57.973275+0200 Power UP![897:87822] uuid: a73e9a10-628f-4494-a099-12efaf72258f value: (null)
2022-09-14 14:42:57.974597+0200 Power UP![897:88380] flutter: valore ricevuto []
2022-09-14 14:42:57.974647+0200 Power UP![897:88380] flutter: Ricevo da txdata: ricevuto <<  >>
2022-09-14 14:42:57.976429+0200 Power UP![897:88380] flutter: valore ricevuto []
2022-09-14 14:42:57.976474+0200 Power UP![897:88380] flutter: Ricevo da txdata: ricevuto <<  >>`