chipweinberger/flutter_blue_plus #49: Checks if Bluetooth functionality

abdelrahman-abied, 2022-04-24:
Checks if Bluetooth functionality not working in android 12 release mode

abdelrahman-abied, 2022-04-26:
problem fixed after add 
`    // FIXME: Work-around for https://github.com/pauldemarco/flutter_blue/issues/772
            shrinkResources false
            minifyEnabled false`

Giuliopretis, 2022-04-27:
I have the same issue, unfortunately this is a workaround and makes my application weight more than it should be. I've read a solution that uses "proguard-rules" to tell the minifier to not shrink to those classes that creates the issue on first instance.

The solution that I've read on flutter_blue GitHub issue is to create "proguard-rules.pro" file and write into it :`-keep class com.boskokg.flutter_blue_plus.* { *; }
-keep class com.boskokg.flutter_blue_plus.Protos* { *; }
-keep class com.boskokg.flutter_blue_plus.Protos.ScanSettings* { *; }`

Then on the build.graddle file under _release_ insert this line `proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'`

Of curse they use another package (com.pauldemarco.flutter_blue) in their solution. I've tried to adapt to the plugin that we use, unfortunately it didn't work.