chipweinberger/flutter_blue_plus #56: Not able to disconnect

tm-topnet, 2022-05-10:
Hi,

i used the library and it works well. 
When i add at the main file this line of code (to not turn the device) the disconnect function doesn't work.

`void main() async {
  WidgetsFlutterBinding.ensureInitialized();
  await SystemChrome.setPreferredOrientations(
    [DeviceOrientation.portraitUp, DeviceOrientation.portraitDown],
  );
  runApp(const MyApp());
}`

I think, that the problem comes becouse the disconnect is a future that doesn't have the async, await state. How can i solve the problem?

janleskovec, 2022-05-17:
The same thing happens to me, although it seems to be limited to iOS.
On Android, I call `dev.disconnect();` and it works fine, however on iOS after calling the function, nothing happens.

zyphlar, 2022-10-20:
Did you ever figure out a resolution? I see a lot of issues mentioning problems with disconnection especially on iOS and they're all randomly getting closed a month-ish later by the originators without any mention of a resolution.