chipweinberger/flutter_blue_plus #58: Throwing platform exception in release mode while starting scan using FlutterBluePlus.instance.scan(timeout:Duration(seconds:30))

vaimikpatel2908, 2022-05-11:
This is the exception I am getting when it calls the FlutterBluePlus.instance.scan(timeout:Duration(seconds:30))

Device: Samsung Galaxy Tab E SM-G973W 
Android OS: 7.1.1
Flutter_Blue_Plus: 1.1.3
Flutter: 2.10.5
```
PlatformException(startScan, Field androidScanMode_ for i.b.a.g0 not found. Known fields are [private int i.b.a.g0.h, private i.c.b.b0$i i.b.a.g0.i, private boolean i.b.a.g0.j, private static final i.b.a.g0 i.b.a.g0.k, private static volatile i.c.b.a1 i.b.a.g0.l], java.lang.RuntimeException: Field androidScanMode_ for i.b.a.g0 not found. Known fields are [private int i.b.a.g0.h, private i.c.b.b0$i i.b.a.g0.i, private boolean i.b.a.g0.j, private static final i.b.a.g0 i.b.a.g0.k, private static volatile i.c.b.a1 i.b.a.g0.l]
```

Havko, 2022-05-18:
Not sure if you found a solution to your issue.  Adding this line to android/app/src/proguard-rules.pro fixed this issue for me.

-keep class com.boskokg.flutter_blue_plus.** { *; }

vaimikpatel2908, 2022-05-18:
Thanks this solved my issue.

closed 2022-05-18 by vaimikpatel2908