chipweinberger/flutter_blue_plus #60: Issue in building android app with flutter blue plus

parag-kulkarni, 2022-05-12:
[flutter_blue_plus_package_error.txt](https://github.com/boskokg/flutter_blue_plus/files/8675042/flutter_blue_plus_package_error.txt)

Facing an exception during buidling android app (due to flutter blue plus). PFA the log for details.
Other system details:
1. Android SDK version: 19
2. JDK version: 8/11/17 (all versions show similar issue)

Tried create a separate project with only flutter_blue_plus package (to isolate the other dependencies). Still facing the same issue.

boskokg, 2022-05-12:
@parag-kulkarni run flutter doctor. it looks like you are not running java 11/17 but java 8:

An exception has occurred in the compiler (1.8.0_242-release)

bufolab, 2022-05-26:
@parag-kulkarni 
In file android/gradle.properties  set this line:
`org.gradle.java.home=/Library/Java/JavaVirtualMachines/jdk-11.0.1.jdk/Contents/Home`
Note the path to your jdk11 is likely different

closed 2022-05-27 by boskokg