chipweinberger/flutter_blue_plus #61: Bluetooth connection problem

HanJiuYuan, 2022-05-12:
How to judge bluetooth status when the program is still connected after the bluetooth device is manually shut down

Hitbee-dev, 2022-05-19:
``` dart
BluetoothDeviceState event
```
Please check this out

HanJiuYuan, 2022-05-19:
> ```dart
> BluetoothDeviceState event
> ```
> 
> 请检查一下

Can you tell me the specific way to achieve it?

HanJiuYuan, 2022-05-23:
> For example, The code below is included in DeviceScreen.
> 
> [image]
> 
> `print("snapshot.data: ${snapshot.data}");` You can use to get the current state.
> 
> Result: [image]

Thank you. That's very helpful

closed 2022-07-23 by boskokg