chipweinberger/flutter_blue_plus #66: The app stays on the "Bluetooth adapter is unknown" screen with iPhone (iOS 15.4.1)

sercan-uzun, 2022-05-23:
Hello folk. I am devoloping my first app on flutter with flutter_blue_plus.

If I run the application while it is already installed, I'm getting the "Bluetooth Adapter is unknown" error. If I run it again with the flutter run button or delete the application and reinstall it, the problem disappears. I don't see any error log on debug screen.
And It's not permissions issue. Because, if I restart with run button, working very well.

I also find that, this same problem has been experienced before, in the link below. It's look like a bug. How can I solve this problem? Thanks in advance.

https://github.com/pauldemarco/flutter_blue/issues/690

sercan-uzun, 2022-05-23:
By the way I am using example code. Calling Off screen like this:

class FlutterBlueApp extends StatelessWidget {
  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      theme: ThemeData.dark(),
      color: Colors.lightBlue,
      home: StreamBuilder<BluetoothState>(
          stream: FlutterBluePlus.instance.state,
          initialData: BluetoothState.unknown,
          builder: (c, snapshot) {
            final state = snapshot.data;
            if (state == BluetoothState.on) {
              return const FindDevicesScreen();
            }
            return BluetoothOffScreen(state: state!); <-------- this line is calling Off screen with unknown.
          }),
    );
  }
}

arnsteinkjellevold, 2022-05-23:
I have the same issue, and I reported it before "flutter_blue_plus" was created.

I thought I found a workaround by accessing the "isOn" property before checking the state, however this only fixed it during development. Building a release to the users still had the same issue.

This issue started back in "flutter_blue" at the same time RXDart was updated from 0.26.X > 0.27.X so I suspect this is involved in the problem in some way.

arnsteinkjellevold, 2022-05-25:
My app is structured differently using a repository holding the bluettoth instance and a BLOC handling the bluetooth state so i'm not using streambuilder but I guess you could put something like this inside your build just before the return MaterialApp

    final _btInstance = FlutterBluePlus.instance;
    print(_btInstance.isOn.toString());
    sleep(const Duration(milliseconds: 100));

sercan-uzun, 2022-05-30:
> My app is structured differently using a repository holding the bluettoth instance and a BLOC handling the bluetooth state so i'm not using streambuilder but I guess you could put something like [...]

I tried this but it's not worked for me! I actually tried more but I am still looking solution. Thank you for your attention. Do you have another suggestion?

arnsteinkjellevold, 2022-05-30:
You could try to start with a longer timeout, say 3000 ms, and see if it works then.
Really what we want is either state ON or OFF so it might be possible to loop while calling .isOn until you either get  the ON or OFF state.

sercan-uzun, 2022-05-30:
I tried for 1000 ms and not worked. I will try 3000 ms.

dogan-kartal, 2022-06-02:
I have the same issue but when I turn off and turn on bluetooth via device everything works fine.
In my research, problem is library can't recognize IOS bluetooth state.
I am not sure this is the right way but this code work for me:
`if (state == BluetoothState.on || state == BluetoothState.turningOn || state == BluetoothState.unknown) {
              return const FindDevicesScreen();
            }`

arnsteinkjellevold, 2022-06-02:
@dogan-kartal I would not trust those other states i production, particularly not BluetoothState.unknown. It might be that the actual state is BluetoothState.off.

**NOTE: The following is not a recommendation for a workaround but rather some input to try to find a way to fix the issue inside flutter_blue_plus**

@boskokg pease have a look...
I have been digging inside the flutter_blue_plus code a little and I can see that the initial state is read using a method (_channel.invokeMethod('state')), however this returns "unknown" when it is called immediately after the app launches. This seems to be a timing issue in iOS.

Then the actual stream is subscribed to. But being a broadcast stream it will not trigger until there is another subsequent state change. So when the initial state was returned as "unknown" when it was actually "on" this state status until bluetooth is toggled.

What I found was that if I swapped the calls and start listening to the broadcast stream first, and then use the method to get the current state, I get the correct state (on/off) every time. I have tested it in release for both iOS and Android. 

I don't know if listening to the stream first is just by chance taking enough time to get the state correct or if it's forcing the method to get the correct state.

```
/// Gets the current state of the Bluetooth module
  Stream<BluetoothState> get state async* {
    yield* _stateChannel
        .receiveBroadcastStream()
        .map((buffer) => protos.BluetoothState.fromBuffer(buffer))
        .map((s) => BluetoothState.values[s.state.value]);

    yield await _channel
        .invokeMethod('state')
        .then((buffer) => protos.BluetoothState.fromBuffer(buffer))
        .then((s) => BluetoothState.values[s.state.value]);
  }
```

I can also get the correct state if I read the state twice using the method:

```
/// Gets the current state of the Bluetooth module
  Stream<BluetoothState> get state async* {
    yield await _channel
        .invokeMethod('state')
        .then((buffer) => protos.BluetoothState.fromBuffer(buffer))
        .then((s) => BluetoothState.values[s.state.value]);

    yield await _channel
        .invokeMethod('state')
        .then((buffer) => protos.BluetoothState.fromBuffer(buffer))
        .then((s) => BluetoothState.values[s.state.value]);

    yield* _stateChannel
        .receiveBroadcastStream()
        .map((buffer) => protos.BluetoothState.fromBuffer(buffer))
        .map((s) => BluetoothState.values[s.state.value]);
  }
```
...or just adding a delay, but I'm not comfortable with that solution:

```
/// Gets the current state of the Bluetooth module
  Stream<BluetoothState> get state async* {
    
    await Future.delayed(const Duration(milliseconds: 100), () {
      print('Workaround - wait for BT state ...');
    });

    yield await _channel
        .invokeMethod('state')
        .then((buffer) => protos.BluetoothState.fromBuffer(buffer))
        .then((s) => BluetoothState.values[s.state.value]);

    yield* _stateChannel
        .receiveBroadcastStream()
        .map((buffer) => protos.BluetoothState.fromBuffer(buffer))
        .map((s) => BluetoothState.values[s.state.value]);
  }
```

arnsteinkjellevold, 2022-06-14:
My understanding is that bluetooth need be available before we get the initial state.
Here is one suggestion on how to do this. (I'm sure there is better and more elegant ways):

```
/// Gets the current state of the Bluetooth module
  Stream<BluetoothState> get state async* {

    // Wait up to one second for bluetooth to become available.
    var retries = 0;
    while ((retries < 10) & !(await this.isAvailable)) {
      retries++;
      await Future.delayed(Duration(milliseconds: 100));
    }

    yield await _channel
        .invokeMethod('state')
        .then((buffer) => protos.BluetoothState.fromBuffer(buffer))
        .then((s) => BluetoothState.values[s.state.value]);

    yield* _stateChannel
        .receiveBroadcastStream()
        .map((buffer) => protos.BluetoothState.fromBuffer(buffer))
        .map((s) => BluetoothState.values[s.state.value]);
  }
```

arnsteinkjellevold, 2022-06-15:
Just to substantiate my assumption regarding "isAvailable", here is a snippet from the iOS plugin code:

```
else if([@"isAvailable" isEqualToString:call.method]) {
    if(self.centralManager.state != CBManagerStateUnsupported &&
       self.centralManager.state != CBManagerStateUnknown) {
      result(@(YES));
    } else {
      result(@(NO));
    }
```
IsAvailable makes sure that the state is not unknown.

sprimm98, 2022-06-24:
Hi @boskokg, have you already had the time to investigate this? A fix would be appreciated :)

Wackymax, 2022-08-17:
Looks like this issue still exists?

sercan-uzun, 2022-08-17:
Yes it is still exist. I changed BLE library with flutter_reactive_ble

bernier, 2022-12-20:
I don't mean to be pushy, but can we get a fix? It looks like commits and PR have been there since September. Thanks

chipweinberger, 2023-04-27:
merged.  https://github.com/boskokg/flutter_blue_plus/commit/536175eb771f24db88f823c678580cc918f98cda

chipweinberger, 2023-04-27:
I tested it, and it does fix the problem for me.

BonDizard, 2024-07-25:
Issue Description
Did the issue get resolved? For Android, I'm getting the following error:

[image]

We can see that Bluetooth is on, yet I'm getting the error. I just used the example code in the flutter_blue_plus package. For the device off state, I used the following code:

```
class BluetoothOffScreen extends StatelessWidget {
  final BluetoothAdapterState bluetoothAdapterState;

  const BluetoothOffScreen({super.key, required this.bluetoothAdapterState});

  @override
  Widget build(BuildContext context) {
    String message;
    IconData icon;
    Color iconColor;

    switch (bluetoothAdapterState) {
      case BluetoothAdapterState.unknown:
        message = 'Bluetooth Status Unknown';
        icon = Icons.help_outline;
        iconColor = Colors.amber;
        break;
      case BluetoothAdapterState.off:
        message = 'Bluetooth is Disabled';
        icon = Icons.bluetooth_disabled;
        iconColor = Colors.white;
        break;
      case BluetoothAdapterState.on:
        // This shouldn't reach this screen, you can navigate elsewhere.
        Navigator.of(context).pop(); // Remove from navigation stack
        return Container(); // Empty container to prevent rendering issues
      default:
        message = 'Unexpected State';
        icon = Icons.error_outline;
        iconColor = Colors.red;
    }

    return Scaffold(
      backgroundColor: Theme.of(context).colorScheme.error,
      body: Center(
        child: Column(
          mainAxisAlignment: MainAxisAlignment.center,
          crossAxisAlignment: CrossAxisAlignment.center,
          children: [
            Icon(
              icon,
              size: 72.0,
              color: iconColor,
            ),
            const Padding(padding: EdgeInsets.all(16.0)),
            Text(
              message,
              style: const TextStyle(
                fontFamily: 'Alatsi',
                color: Colors.white,
                fontSize: 30,
              ),
            ),
            const Padding(padding: EdgeInsets.all(8.0)),
            const Text(
              'Please enable Bluetooth to connect to devices.',
              style: TextStyle(
                fontFamily: 'Alatsi',
                color: ColorConstants.darkColor,
              ),
            ),
            const Padding(padding: EdgeInsets.all(16.0)),
            ReusableButton(
              onPressed: () async {
                await requestBluetoothPermission();
              },
              buttonText: 'Turn on Bluetooth',
            ),
          ],
        ),
      ),
    );
  }

  Future<void> requestBluetoothPermission() async {
    try {
      if (Platform.isAndroid) {
        await FlutterBluePlus.turnOn();
      }
    } catch (e) {
      CustomToast.showToast("Error Turning On: $e");
    }
  }
}
```

The below code is my main.dart. It's exactly as the example code, but still, I'm sending it if it helps:
```
void main() {
  FlutterBluePlus.setLogLevel(LogLevel.verbose, color: true);
  WidgetsFlutterBinding.ensureInitialized();
  runApp(
    const ProviderScope(
      child: FlutterBlueApp(),
    ),
  );
}

class FlutterBlueApp extends StatefulWidget {
  const FlutterBlueApp({Key? key}) : super(key: key);

  @override
  State<FlutterBlueApp> createState() => _FlutterBlueAppState();
}

class _FlutterBlueAppState extends State<FlutterBlueApp> {
  BluetoothAdapterState _adapterState = BluetoothAdapterState.unknown;

  late StreamSubscription<BluetoothAdapterState> _adapterStateStateSubscription;

  @override
  void initState() {
    super.initState();
    _adapterStateStateSubscription =
        FlutterBluePlus.adapterState.listen((state) {
      _adapterState = state;

      if (mounted) {
        setState(() {});
      }
      if (!mounted) return;
    });
  }

  @override
  void dispose() {
    _adapterStateStateSubscription.cancel();
    super.dispose();
  }

  @override
  Widget build(BuildContext context) {
    Widget screen;
    if (_adapterState == BluetoothAdapterState.on
        // || _adapterState == BluetoothAdapterState.turningOn
        // || _adapterState == BluetoothAdapterState.unknown
        ) {
      screen = const ScanPage();
    } else {
      screen = BluetoothOffScreen(
        bluetoothAdapterState: _adapterState,
      );
    }

    return MaterialApp(
      debugShowCheckedModeBanner: false,
      home: screen,
      navigatorObservers: [BluetoothAdapterStateObserver()],
    );
  }
}
```

Below is my flutter doctor output if it helps:
`
flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[√] Flutter (Channel stable, 3.22.2, on Microsoft Windows [Version 10.0.22621.3007], locale en-IN)
[√] Windows Version (Installed version of Windows is version 10 or higher)
[√] Android toolchain - develop for Android devices (Android SDK version 34.0.0)
[√] Chrome - develop for the web
[√] Visual Studio - develop Windows apps (Visual Studio Community 2022 17.8.4)
[√] Android Studio (version 2024.1)
[√] VS Code (version 1.91.0)
[√] Connected device (4 available)
[√] Network resources

• No issues found!
`

One interesting thing is that in Moto Edge 40, it's working, but in Vivo Z1 and Redmi 10 Prime, I'm getting the issue.

sampaio96, 2025-02-27:
I too have this problem.