chipweinberger/flutter_blue_plus #67: Device is not able to discover in app but showing in mobile's bluetooth settings

nbsathawara, 2022-05-25:
Hi,

I have checked using given example as well as integrating in my app but I am not able to get particular device in scan result
in any android OS (Checked with 7,9,11,12).

**But device is showing in mobile's bluetooth settings as available device.**

I have also checked using native android development just to make sure and device is showing in scan result. 

Please refer to the attached images. 

Device name : OS_Horti_PRM_2

1st image is from mobile's bluetooth settings

[image]

 and 2nd is from my app.

[image]

Any help would be appreciated.

Thanks
Nikhil Sathawara

arnsteinkjellevold, 2022-05-25:
Is it a BLE device?

nbsathawara, 2022-05-26:
@arnsteinkjellevold  I thought so but from the scan result looks like its not.
It may be classic bluetooth device.

nbsathawara, 2022-05-27:
Hello Guys,
 
I have find out that the target device is running on classic bluetooth and not BLE.
So, will look into classic bluetooth libraries.

closed 2022-05-27 by nbsathawara