chipweinberger/flutter_blue_plus #68: Error in write and notify characteristics platform exception

anas-iqbal, 2022-05-25:
Hello,

I have been facing the same issue with both flutter blue and flutter blue plus that is sometimes this below line starts giving exception.

```
await setNotifyValue(true); // exception: set_notification_error, error when writing the descriptor, null, null

await writeChar.write(cmd); // exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null).
```

I am not running this lines concurrently I have used await. I am working with 3 BLE devices,  it works most of them time but sometimes it start getting this exception and my app misbehaves and this error can happen in any of the device not even specific to any device. any idea?

designerslab, 2022-06-02:
It happens when characteristics are not found.
Use State management Bloc or Provider 
or try with the Stream widget.

anas-iqbal, 2022-06-03:
Hi designerslab

thanks for the response but I really don't think this is the problem here, because I also have a null check on both above lines. What I have been able to found out so far that scan results gives same devices multiple times and when these functions are executed multiple times then this error occurs.

BellRampion, 2022-10-28:
I'm having the same issue. The characteristics are definitely found and the device receives data, but I still get the "write_characteristic_error".

anas-iqbal, 2022-10-29:
Hey @BellRampion I don't know how you are implementing it, but for me after moving all this bluetooth code into singleton class helped, what I found so far is that sometime even after closing the bluetooth connection it doesn't get actually closed, thats when this error happens or better solution is just to write native code for bluetooth.

chipweinberger, 2023-07-20:
same issue: https://github.com/boskokg/flutter_blue_plus/issues/385

Please try 1.7.6+. reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger