chipweinberger/flutter_blue_plus #69: Why is the iOS reconnection function removed from the latest version 1.3.1?

Superyanggod, 2022-05-29:
Why is the iOS reconnection function removed from the latest version 1.3.1?

sophisticode, 2022-05-30:
It brought unwanted behaviour for many use-cases. Please check https://github.com/boskokg/flutter_blue_plus/pull/11

closed 2022-06-03 by boskokg

Superyanggod, 2022-06-08:
> It brought unwanted behaviour for many use-cases. Please check #11
So, if iOS needs to automatically reconnect after disconnecting, how to achieve it?
Duplicate of #76

sophisticode, 2022-06-09:
Unfortunately, iOS does not have an auto-connect option as Android has.
So if a BLE device goes out of range and come backs into range later, there is no way the OS will connect automatically again by itself. On iOS, it will be necessary to do a scan, and if the desired device is found in the scan results, the app can connect to it again.

The change that was undone in https://github.com/boskokg/flutter_blue_plus/pull/11 did not implement an auto-connect as Android does. It simply tried to connect whenever a disconnect was issued by BLE device (while still being in range). You can achieve the same behaviour in your app with a code like this:

```
  bool disconnectByUser = false;

  // Call this code for user initiated disconnects
  userDisconnect(BluetoothDevice peripheral) async {
    disconnectByUser = true;
    await peripheral.disconnect();
    await Future.delayed(Duration.zero); // To be sure the disconnect callback is done before changing boolean
    disconnectByUser = false;
}

  // Place this code where you listen to the peripheral state
  peripheral.state.listen((connectionState) {
    if (connectionState == BluetoothDeviceState.disconnected) {
      if (!disconnectByUser) {
        peripheral.connect();
      }
    }
  });
}

```

sophisticode, 2022-06-09:
Nevertheless, we could think of having some native code doing it on iOS as https://github.com/boskokg/flutter_blue_plus/commit/c7bbdf0c8dbf808c5081a8db8ed00c4eb0bce4ef, but only depending on a flag that could be set in the connect() method.

Superyanggod, 2022-08-25:
> 尽管如此，我们可以考虑在 iOS 上使用一些本地代码作为[c7bbdf0](https://github.com/boskokg/flutter_blue_plus/commit/c7bbdf0c8dbf808c5081a8db8ed00c4eb0bce4ef)执行此操作，但仅取决于可以在 connect() 方法中设置的标志。

Hi, so far I have done what you said, but I don't think it's a good solution because I have to update the bluetooth function every time I connect, don't know if there is a better solution, most umm , can be implemented on the native side, just leave the method on the Flutter side.