chipweinberger/flutter_blue_plus #70: Crash on Android 12

mit-73, 2022-05-30:
The crash happens after the bluetooth and geo position are enabled. In debug mode, it does not replay, the plugin works fine. In release it crashes like this.

**ProtoMaker.java line 47:**
```
Fatal Exception: java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1452, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity {com.example/com.example.MainActivity}: java.lang.RuntimeException: Field remoteId_ for e.a.a.o not found. Known fields are [private static final e.a.a.o e.a.a.o.p, private static volatile e.b.c.a1 e.a.a.o.q]
       at android.app.ActivityThread.deliverResults(ActivityThread.java:5422)
       at android.app.ActivityThread.handleSendResult(ActivityThread.java:5461)
       at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:54)
       at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
       at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
       at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2259)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loopOnce(Looper.java:210)
       at android.os.Looper.loop(Looper.java:299)
       at android.app.ActivityThread.main(ActivityThread.java:8105)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:556)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1045)
```

```
Caused by java.lang.RuntimeException: Field remoteId_ for e.a.a.o not found. Known fields are [private static final e.a.a.o e.a.a.o.p, private static volatile e.b.c.a1 e.a.a.o.q]
       at com.google.protobuf.MessageSchema.reflectField(MessageSchema.java:72)
       at com.google.protobuf.MessageSchema.newSchemaForRawMessageInfo(MessageSchema.java:655)
       at com.google.protobuf.MessageSchema.newSchema(MessageSchema.java:12)
       at com.google.protobuf.ManifestSchemaFactory.newSchema(ManifestSchemaFactory.java:60)
       at com.google.protobuf.ManifestSchemaFactory.createSchema(ManifestSchemaFactory.java:49)
       at com.google.protobuf.Protobuf.schemaFor(Protobuf.java:17)
       at com.google.protobuf.Protobuf.schemaFor(Protobuf.java:4)
       at com.google.protobuf.GeneratedMessageLite.makeImmutable(GeneratedMessageLite.java:4)
       at com.google.protobuf.GeneratedMessageLite$Builder.buildPartial(GeneratedMessageLite.java:9)
       at com.google.protobuf.GeneratedMessageLite$Builder.build(GeneratedMessageLite.java)
       at com.boskokg.flutter_blue_plus.ProtoMaker.from(ProtoMaker.java:47)
       at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$2(FlutterBluePlusPlugin.java:48)
       at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$2$FlutterBluePlusPlugin(FlutterBluePlusPlugin.java)
       at com.boskokg.flutter_blue_plus.-$$Lambda$FlutterBluePlusPlugin$Oj29_kJ5l8Z7G46y2EV75FHeJT0.op(-.java:4)
       at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.lambda$ensurePermissionBeforeAction$4(FlutterBluePlusPlugin.java:11)
       at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.lambda$ensurePermissionBeforeAction$4$FlutterBluePlusPlugin(FlutterBluePlusPlugin.java)
       at com.boskokg.flutter_blue_plus.-$$Lambda$FlutterBluePlusPlugin$n-o5LNC3hA6GoJcqGVOBTBvOoxg.op(-.java:4)
       at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onRequestPermissionsResult(FlutterBluePlusPlugin.java:28)
       at io.flutter.embedding.engine.FlutterEngineConnectionRegistry$FlutterEngineActivityPluginBinding.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:20)
       at io.flutter.embedding.engine.FlutterEngineConnectionRegistry.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:20)
       at io.flutter.embedding.android.FlutterActivityAndFragmentDelegate.onRequestPermissionsResult(FlutterActivityAndFragmentDelegate.java:59)
       at io.flutter.embedding.android.FlutterActivity.onRequestPermissionsResult(FlutterActivity.java:10)
       at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:8781)
       at android.app.Activity.dispatchActivityResult(Activity.java:8634)
       at android.app.ActivityThread.deliverResults(ActivityThread.java:5415)
       at android.app.ActivityThread.handleSendResult(ActivityThread.java:5461)
       at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:54)
       at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
       at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
       at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2259)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loopOnce(Looper.java:210)
       at android.os.Looper.loop(Looper.java:299)
       at android.app.ActivityThread.main(ActivityThread.java:8105)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:556)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1045)
```

**FlutterBluePlugin.java line 56:**
```
Fatal Exception: java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1452, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity {com.example/com.example.MainActivity}: java.lang.RuntimeException: Field devices_ for e.c.a.q not found. Known fields are [private e.a.c.b0$i e.c.a.q.p, private static final e.c.a.q e.c.a.q.q, private static volatile e.a.c.a1 e.c.a.q.r]
       at android.app.ActivityThread.deliverResults(ActivityThread.java:5422)
       at android.app.ActivityThread.handleSendResult(ActivityThread.java:5461)
       at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:54)
       at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
[... 27 lines cut ...]
       at com.pauldemarco.flutter_blue.FlutterBluePlugin.lambda$ensurePermissionsBeforeAction$3(FlutterBluePlugin.java:11)
       at com.pauldemarco.flutter_blue.FlutterBluePlugin.lambda$ensurePermissionsBeforeAction$3$FlutterBluePlugin(FlutterBluePlugin.java)
       at com.pauldemarco.flutter_blue.-$$Lambda$FlutterBluePlugin$NJkGjHDzsWKb5egoEZiOXPik7EM.op(-.java:4)
       at com.pauldemarco.flutter_blue.FlutterBluePlugin.onRequestPermissionsResult(FlutterBluePlugin.java:28)
       at io.flutter.embedding.engine.FlutterEngineConnectionRegistry$FlutterEngineActivityPluginBinding.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:20)
       at io.flutter.embedding.engine.FlutterEngineConnectionRegistry.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:20)
       at io.flutter.embedding.android.FlutterActivityAndFragmentDelegate.onRequestPermissionsResult(FlutterActivityAndFragmentDelegate.java:59)
       at io.flutter.embedding.android.FlutterActivity.onRequestPermissionsResult(FlutterActivity.java:10)
       at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:8781)
       at android.app.Activity.dispatchActivityResult(Activity.java:8634)
       at android.app.ActivityThread.deliverResults(ActivityThread.java:5415)
       at android.app.ActivityThread.handleSendResult(ActivityThread.java:5461)
       at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:54)
       at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
       at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
       at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2259)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loopOnce(Looper.java:210)
       at android.os.Looper.loop(Looper.java:299)
       at android.app.ActivityThread.main(ActivityThread.java:8105)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:556)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1045)
```

arnsteinkjellevold, 2022-05-30:
Why are the error referencing com.pauldemarco.flutter_blue.FlutterBluePlugin?
Should it not be com.boskokg.flutter_blue_plus?

mit-73, 2022-05-30:
@arnsteinkjellevold Sorry didn't notice, I'm testing several plugins at once. I left the information related to com.boskokg.flutter_blue_plus.

arnsteinkjellevold, 2022-06-03:
In proguards-rules.pro I'm using this to get permissions to work:
-keep class com.boskokg.flutter_blue_plus.Protos* { *; }

If the issue related to BluetoothGatt is not directly related to the original issue then then I would suggest to close this issue and create a new issue for BluetoothGatt and document the error you get there.

closed 2022-06-03 by boskokg