chipweinberger/flutter_blue_plus #74: Operations in background

flerik12, 2022-06-01:
Hello
I am a little confused how to proprely use this package in background instead of performing all the tasks right after user clicks on the screen.
I have an app in which all the user have to do is to pick the desired device from device list. Then he is monitoring all the data on UI, and he is not able to perform any tasks, he is able just to view the parameters from the device.

In the example, the user choose device -> start searching for services -> the buildServiceTiles builds the UI for him using StreamBuilders -> he choose some characteristics and perform tasks on them.

In my case, the user choose device -> the app does everything above in the backgroung (search for services, perform periodical tasks on characteristics to receive neccessary data for the UI, process the data and display it).

I am curious if there is a quick and easy way to achieve this? Did anyone had similar task?

sprimm98, 2022-06-24:
I'm using flutter_background (https://pub.dev/packages/flutter_background) to keep the bluetooth connection running, processing and storing data and doing HTTP requests periodically. It works great on Android, but I don't know if there is a way to achieve that behavior on iOS.

fsievers, 2022-07-05:
I'm currently using `flutter_foreground_task` on Android and iOS to do periodic background BLE scans and it works greate. On iOS you have to make sure, that your application is allowed to use bluetooth in background-mode (in xcode add 'Uses Bluetooth LE accessories' on the 'Signing & Capabilities' page for you app).

closed 2022-07-23 by boskokg

fabiototh, 2022-09-21:
@fsievers could you show me a example?

jawadabbasi923, 2023-02-27:
@flerik12  Did you solved it ?