chipweinberger/flutter_blue_plus #84: Unexpected reconnect after being disconnected

sprimm98, 2022-06-24:
Hi! I'm working on an app that monitors incoming bluetooth data and uses the bluetooth device state stream to detect if the peripheral gets disconnected due to getting out of range. If it does, the app returns back to the main menu. Basically, it works great, but sometimes the device reconnects to the peripheral again although autoConnect is set to false. Is the reconnection an expected behavior and if so, how can I avoid that? Thanks.

[Device: Pixel 2 XL, Android 11]

alericciuto, 2022-07-03:
same issue here

boskokg, 2022-07-24:
@sprimm98 @alericciuto  I cannot reproduce this issue. Please make an investigation and give us the merge reqeust proposal for the fix.
Thanks.

closed 2022-07-24 by boskokg