chipweinberger/flutter_blue_plus #87: iOS every 30 seconds i get (didDisconnectPeripheral)

simondukaric, 2022-06-29:
Only on iOS devices every 30 seconds i get (didDisconnectPeripheral)... Android works fine. Any solution to this or whats causing this?

Ladegeraet, 2022-07-12:
I have the same problem. Although not every 30 seconds and not on every iPhone, but the error occurs very often. 
@simondukaric have you already found a solution?

simondukaric, 2022-07-12:
In a "hacky" way yes. I have stored device in state and I listen to device connection state stream, and when state is "disconnected" i immediately connect again to the saved device. But this wouldnt work if you actually turn off BLE device or go outside of the proximity of the device...

boskokg, 2022-07-24:
@simondukaric @Ladegeraet I cannot reproduce this issue.
Please make investigation and give us the merge request to fix it.

closed 2022-07-24 by boskokg