chipweinberger/flutter_blue_plus #90: Android: unable to scan for BLE beacons

fsievers, 2022-07-07:
I want to switch from `flutter_blue` to `flutter_blue_plus` but found a problem with the `startScan()` method and the results on Android (iOS is not affected).

First starting a scan with the option `withServices: [Guid("0000feaa-0000-1000-8000-00805f9b34fb")]` does not find any beacons (it works with `flutter_blue`). The UUID is for Eddystone Beacons.

If I don't use the `withServices` option for `startScan`, I'm able to find lots of devices but none of them has the Eddystone ServiceUUID. For example this is a result for a known beacon with `flutter_blue_plus`:  
Format: `(MAC: [ScanResult.advertisementData.serviceUuids] {ScanResult.advertisementData.serviceData})`  
```
AC:23:3F:E6:4A:D1: [0000ffe1-0000-1000-8000-00805f9b34fb] {0000ffe1-0000-1000-8000-00805f9b34fb: [161, 8, 100, 209, 74, 230, 63, 35, 172, 80, 76, 85, 83]}
```

The following is how it should look like and how it looks with `flutter_blue` (Beacons sends Eddystone UID, URL and TLM):
```
AC:23:3F:E6:4A:D1: [0000feaa-0000-1000-8000-00805f9b34fb] {0000feaa-0000-1000-8000-00805f9b34fb: [0, 232, 222, 173, 190, 239, 222, 173, 190, 239, 222, 173, 222, 173, 190, 239, 0, 3]}
AC:23:3F:E6:4A:D1: [0000feaa-0000-1000-8000-00805f9b34fb] {0000feaa-0000-1000-8000-00805f9b34fb: [16, 232, 3, 116, 101, 108, 101, 115, 116, 101, 46, 115, 121, 115, 116, 101, 109, 115]}
AC:23:3F:E6:4A:D1: [0000feaa-0000-1000-8000-00805f9b34fb] {0000feaa-0000-1000-8000-00805f9b34fb: [32, 0, 12, 144, 21, 0, 1, 168, 24, 120, 5, 10, 92, 98]}
```

Here is a minimal excerp from my code:
```dart
FlutterBluePlus ble = FlutterBluePlus.instance;
// …
// left out code to check if ble is available and some other stuff not related to the scan
// …
ble.scanResults.listen((results) async {
    for (ScanResult r in results) {
        final serviceData = r.advertisementData.serviceData;
        final serviceUuids = r.advertisementData.serviceUuids;

        log('${r.device.id}: $serviceUuids $serviceData');
    }
});

await ble.startScan(
    timeout: const Duration(seconds: 5),
    //withServices: [Guid("0000feaa-0000-1000-8000-00805f9b34fb")],
);
```

blacktorch, 2022-07-18:
This issue is due to the android permission flag put in the AndroidManifest file in the plugin flutter_blue_plus/android/src/main/AndroidManifest.xml. IBeacons are used for location applications most times, and 
android:usesPermissionFlags="neverForLocation" makes the android system filter out Beacon devices during scan, so by removing this from the permission,
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
and have this instead,
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
The issue will be solved.

fsievers, 2022-07-18:
@blacktorch thank you, that did the trick. I changed the line in the AndroidManifest.xml in the pub-cache directory for flutter_blue_plus. But this is just a hack for a quick test. Is there a way to override the line or the file from within my project directory, without doing a fork of flutter_blue_plus?

closed 2022-07-18 by fsievers