chipweinberger/flutter_blue_plus #92: iOS is parsing manufacturerData incorrectly

sqcsabbey, 2022-07-14:
For the same peripheral, I get different AdvertisementData.manufacturerData; for the same manufacturer key, here are the values I'm seeing:

```
Android: [91, 1, 1, 0, 0, 0, 0, 0, 0]
iOS:  [0, 91, 1, 1, 0, 0, 0, 0, 0, 0]
```

... there's an extra leading 0 for iOS. From the manufacturer's documentation, the Android version is the correct one.

boskokg, 2022-07-24:
@sqcsabbey thanks. Please send a merge request to fix this issue.

sqcsabbey, 2022-07-26:
@boskokg I submitted the pull request, please let me know if there's more I can do to help with this one, and thank you for maintaining this project!

closed 2022-08-04 by boskokg

Time-L, 2022-08-09:
For the same peripheral, I get different AdvertisementData.manufacturerData; for the same manufacturer key, here are the values I'm seeing:

Android:[4, 22, 6, 0, 80]
iOS: [22, 6, 0, 80]

IOS lost the first value. From the manufacturer's documentation, the Android version is the correct one.

sqcsabbey, 2022-08-09:
My fix made my case work, but it doesn't look good; @boskokg — please undo my MR. 

I just set myself up for Flutter plugin development and I'm going to figure out exactly what's going on with my particular case.

sqcsabbey, 2022-08-10:
I'm still looking into this... 

My peripheral is sending these bytes:

0x0c 0xff 0xe5 0x02 0x00 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00

The 0c means there are 12 more bytes coming.
The ff means it's a manufacturer message. 
The next 2 bytes are the manufacturer id.
The next 9 bytes are the custom manufacturer data.

Here's what I'm getting in this plugin's toScanResultProto call, called from handling CBCentralManager.didDiscoverPeripheral:

  advertisementData: {
      kCBAdvDataIsConnectable = 1;
      kCBAdvDataLocalName = "SQ-SPT-0021 432190009";
      kCBAdvDataManufacturerData = {length = 12, bytes = 0xe50200000101000000000000};
      kCBAdvDataRxPrimaryPHY = 0;
      kCBAdvDataRxSecondaryPHY = 0;
      kCBAdvDataServiceUUIDs =     (
          "00000100-34ED-12EF-63F4-317792041D17"
      );
      kCBAdvDataTimestamp = "681852653.472335";
  }

... it looks like Apple is adding an extra byte to the kCBAdvDataManufacturerData value? I'm trying to track down why that might be... there's [this post](https://developer.apple.com/forums/thread/19060):

> What you are seeing is due to a change in iOS 9.
>
> If the accessory advertises mfg data in both the advertising packet (ADV_IND) and the scan response (SCAN_RSP), previously iOS would only hand back the data from the scan response. iOS 9 will now concatenate the two fields to provide all the data from the device.

... that probably isn't my particular issue, but there's something going on...

sqcsabbey, 2022-08-11:
I'm stuck on this one, asked [here](https://developer.apple.com/forums/thread/712142).

Luzzotica, 2022-08-17:
My manufacturers data on iOS is also being truncated, losing the first value.
Can we revert the merge request done by sqcsabbey?

PR #104 (2022-08-04): fixed manufacturer data parsing
Fixes  #92.