chipweinberger/flutter_blue_plus #96: Incorrect Characteristics Notification Data When Reconnecting

FXschwartz, 2022-07-20:
This is an [issue](https://github.com/pauldemarco/flutter_blue/issues/525) that existed on the original flutter_blue library, and I've confirmed I see the same issue here.

For me, it only happens on iOS, and when I pair again to a BLE device, I get duplicate data back, which causes a bunch of issues.

The only way I've been able to fix this is by forking the library and making a change to the bluetooth_characteristics.dart file, commenting out the `_value.stream` line.

This workaround was proposed in the original issue [here](https://github.com/pauldemarco/flutter_blue/issues/525#issuecomment-736671633).

There is also a second issue that was created in the original repo from not too long ago where a developer seems to have a proposed fix for it. That issue is [here](https://github.com/pauldemarco/flutter_blue/issues/1009#issuecomment-1058834600)

boskokg, 2022-07-24:
@FXschwartz thanks. Please send a merge request to fix this issue.

mininglamp-frontend, 2022-08-24:
I also encountered the problem of receiving repeated data on iOS.
Such as:
1. Multiple Characteristics data will also receive confusion.
2. Repeatedly receive the Characteristics of data

Hopefully this can be fixed.

thanks.

ruihua-code, 2022-08-24:
I also encountered the problem of receiving repeated data on iOS.

Such as:
1.Multiple Characteristics data will also receive confusion.
2.Repeatedly receive the Characteristics of data

Hopefully this can be fixed.

thanks.

letisoft, 2023-01-16:
Hello,

I had the same issue and same fix worked for me  - thanks to FXschwartz !!!
Is it possible same to be applied on git?

Thanks!

alexvnesta, 2023-03-12:
I am bumping this issue as it has occurred to me and I am not implementing the above fixes.

chipweinberger, 2023-07-20:
you can use `BluetoothCharacteristic.onValueChangedStream` which does not have the `_value.stream` line.

Please try 1.7.6+. reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger