chipweinberger/flutter_blue_plus #110: Send commands to multiple device and receive messages separately

andres3108, 2022-08-10:
Hello all, I was wondering if you could help me with a question I have: 

I know that is possible to connect to multiple devices at the same time, but now I'm wondering if it is possible to send different commands to these devices and receive their responses separately. 

I have tried doing this with no success, I'm using a class that holds the BluetoothDevice and its characteristics but usually, just one of the connected devices receives and responds.  Worth mentioning that these bluetooth devices are the same brand, so they share the same services and characteristics, the only thing different between them is the device UUID.

Is doing this possible?. Thanks in advance

FXschwartz, 2022-08-10:
This is definitely possible and is probably just due to how you have your classes set up.

For instance, I have an app that connects to various sensors, so I have a class created for each sensor that gets scanned and in that class is a BluetoothDevice property. So each class is a unique Bluetooth sensor.

andres3108, 2022-08-10:
Thanks very much for your answer, I'll try to rework my classes and see where I could go wrong.

closed 2022-08-11 by boskokg