chipweinberger/flutter_blue_plus #113: The first scan is invalid, and the device in the connection list reports an error

xiepeng-xp, 2022-08-16:
1. When the APP is launched for the first time to scan the device, the first scan is invalid, and the second scan is valid.
2. An error occurs when launching the app for the first time and connecting to a device in the list, error: PlatformException(connect, Peripheral not found, null, null)       #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:607:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:167:18)
#2      BluetoothDevice.connect (package:flutter_blue_plus/src/bluetooth_device.dart:39:5)

xiepeng-xp, 2022-08-16:
Supplementary note: everything is normal on the Android platform, and the above error occurs on iOS

mevluttosun, 2022-09-13:
Did you solve the issue? I have the same and I couldn't solve with any of [this](https://github.com/pauldemarco/flutter_blue/issues/497).

closed 2022-12-01 by boskokg