chipweinberger/flutter_blue_plus #115: Where can I get information on how to use this plugin.

davidAg9, 2022-08-19:
I am a fairly new to bluetooth technology , I will like to connect to other bluetooth mobile device ask them about about some info and also send data to them so they do something on that data . How do I achieve this and , I would like someone to point me to documentation or any sort of resource I could use.Thank you.

Silhouette2022, 2022-08-23:
[566988](https://pub.dev/packages/flutter_blue_plus)

Silhouette2022, 2022-08-23:
> I am a fairly new to bluetooth technology , I will like to connect to other bluetooth mobile device ask them about about some info and also send data to them so they do something on that data . How [...]

https://pub.dev/packages/flutter_blue_plus

davidAg9, 2022-08-25:
I have no idea what any of the terminologies  mean...such MTU , characteristics , descriptor ... is there anywhere I can find out where to read on these terminologies and what they are .

Silhouette2022, 2022-08-25:
> 
https://learn.adafruit.com/introduction-to-bluetooth-low-energy?view=all
https://www.google.com/search?q=bluetooth+low+energy+tutorial

closed 2022-12-01 by boskokg