chipweinberger/flutter_blue_plus #120: in Android 12 some beacon devices can not be found

erahani, 2022-08-27:
in Android 12 some beacon devices can not be found using this package.
I checked ScanResult device.name and it is empty for many scanned devices on Android 12. it works on iOS.
I also tried with some third-part apps like light blue and BLE scanner.  BLE scanner can not find it (while it can find same beacons in iOS app). light blue can find it on Android 12 and ios. I appreciate for any help.

navaronbracke, 2022-08-27:
The fact that some beacons can not be found on Android 12 is mentioned in the documentation here:
https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#assert-never-for-location

```
Note: If you include neverForLocation in your android:usesPermissionFlags, some BLE beacons are filtered from the scan results.
```

The plugin does set `neverForLocation` in its manifest.

erahani, 2022-08-31:
Thanks for your reply:
i added this line but did not help. then I manually changed Location App permission to "Allow.." and it worked. not sure how I can avoid manual set of location.
<uses-permission android:name="android.permission.BLUETOOTH_SCAN"
        tools:remove="android:usesPermissionFlags"/>

PR #136 (2022-10-14): Opt out of neverForLocation on Android
This PR opts out of the `neverForLocation` permission flag on Android.

In general having to opt-in is better than having to explicitly opt-out.

Fixes: #135 Fixes: #120
closed 2022-10-14 by boskokg