chipweinberger/flutter_blue_plus #123: iOS disconnection after 1 message

chetan-SMRI, 2022-08-30:
The iOS code connects successfully with ESP32 and sends messages successfully in android without any auto-disconnection

In iOS it gets disconnected immediately after sending the first message.
```
  writeData(String data) async {
    if (targetCharacteristic == null) return;

    List<int> bytes = utf8.encode(data);
    try {await targetCharacteristic.write(bytes);}
    on Exception catch(e){
      print(e);
    }
  }
```

I sent first message and then it also has a wait for 3 sec. but it still cant send a new message,upon checking log it doesn't show much of info regarding why it happened.

closed 2023-03-16 by boskokg

Brinfotech1407, 2023-08-23:
Any update on this as same thing happened for me as I was able to write one message but after that on connection listener received disconnected status of device.

chetan-SMRI, 2023-08-23:
Yes, I remember some details of the project.
It was because the byte length supported by iOS is small rather than Android.
I had resolved it. If u will need, I will try to find exact code and send you. But as far as I remeber, I guess I had to shorter the text I was sending

Brinfotech1407, 2023-08-23:
Thanks @chetan-SMRI for quick response I was looking actually readme where writeLarge extension method is mentioned  if you are referencing to that function then it is fine , otherwise it would be helpful if you send the code , thanks.