chipweinberger/flutter_blue_plus #126: [urgent]IOS manufacturer  scan_ RSP data error!

lybgo, 2022-09-01:
flutter_blue_plus V1.3.1
In IOS ScanResult.advertisementData.manufacturerData value:
{44718: [88, 119, 179, 51, 250, 238, 1, 0, 0, 9, 0, 3, 4, 0, 0, ***174, 175, 2, 0, 6, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 0, 1]***}

In Android value is right:
{44718: [88, 119, 179, 51, 250, 238, 1, 0, 0, 9, 0, 3, 4, 0, 0], ***44974: [2, 0, 6, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 0, 1]***}

the value: "44974: [2, 0, 6, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 0, 1]" is scan_ RSP data 

Help fix this bug

unrealsmart, 2022-10-08:
I don't quite understand. your data is correct. you need to convert according to different platforms!

lybgo, 2022-10-11:
ScanResult.advertisementData.manufacturerData value log print:
Android log:
{<font color=red>44718: [88, 119, 179, 51, 250, 238, 1, 0, 0, 9, 0, 3, 4, 0, 0], </font><font color=blue>44974:[2, 0, 6, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 0, 1]</font>}
red font data is ble scan_AD data, blue font data is scan_ RSP data.
IOS log:
{44718: [88, 119, 179, 51, 250, 238, 1, 0, 0, 9, 0, 3, 4, 0, 0, <font color=blue>174, 175, 2, 0, 6, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 0, 1]</font>}
 
in ios scan_ RSP data error add in  scan_AD data.

closed 2023-03-16 by boskokg