chipweinberger/flutter_blue_plus #127: Is it possible to set CONNECTION PRIORITY?

LathDevers, 2022-09-02:
On some Android devices, we have a **low sampling rate** (connection interval). This leads to sampling rates of approx 20Hz. On other devices, however, it is higher (approx what is given by slave). We figured, that this could possibly be due to the fact that different Android devices automatically balance the connection priority. It would be great if the **connection priority** could be set to high manually.

Is it possible to set the [**CONNECTION_PRIORITY_HIGH**](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#CONNECTION_PRIORITY_HIGH) flag constant in **android.bluetooth.BluetoothGatt**?

*People with similar problem: [Stackoverflow](https://stackoverflow.com/questions/21398766/android-ble-connection-time-interval)*

corteXnzt, 2022-09-04:
Hi I have been using the flutter blue recently and changed to this plus. As I remember I stumbled upon this particular issue before while solving how to make the ble comms faster. The original flutter blue don't have that function AFAIK.  But you can wait for the author's response if any. 

If you are the developer of the embedded device you are working at you can set to request min and max interval to be change to lower value. I tried up to 10ms() data interval with a nrf52832 ble chip. I confirmed it via a bluetooth sniffer but the android device graphing has substantial delay. I'm not certain yet if this library is causing the delay on graph or the graph itself.

Felipev94, 2022-09-25:
Hi ✋, 
I just released a PR #141 with the implementation to make priority connection requests.

chipweinberger, 2023-04-27:
merged here: https://github.com/boskokg/flutter_blue_plus/commit/47b80042a90258cb071c8d1378f939798d9ba612

closed 2023-04-27 by chipweinberger