chipweinberger/flutter_blue_plus #131: connectedDevices return empty List

AhmedAbogameel, 2022-09-05:
I'm calling this code to get connectedDevices 

`print("connectedDevices ${await flutterBlue.connectedDevices}");`

response is an empty array on android and ios
`flutter: connectedDevices []`

narayanvyas, 2022-10-28:
I am also facing a similar kind of issue. In my case, I have to call connectedDevices two times after a 1-second delay, and then it works most of the time as usual. Don't know how to resolve this.

moazelsawaf, 2022-12-03:
@AhmedAbogameel 
Have you found a solution for this one? I am facing the same issue ...

Villian95, 2023-03-01:
I also have the same question,  have you found a solution for this one?

chipweinberger, 2023-07-20:
sounds similar to https://github.com/boskokg/flutter_blue_plus/issues/319

please try 1.7.6+, feel free to reopen if you still have problems.

closed 2023-07-20 by chipweinberger