chipweinberger/flutter_blue_plus #132: write_characteristic_error, writeCharacteristic failed, null, null

yu622311, 2022-09-08:
E/FrameEvents(29528): updateAcquireFence: Did not find frame.
D/FlutterBluePlugin(29528): mDevices size: 1
D/BluetoothGatt(29528): onConnectionUpdated() - Device=D6:E4:7B:08:5C:E8 interval=30 latency=0 timeout=500 status=0
D/BluetoothGatt(29528): discoverServices() - device: D6:E4:7B:08:5C:E8
D/BluetoothGatt(29528): onSearchComplete() = Device=D6:E4:7B:08:5C:E8 Status=0
D/FlutterBluePlugin(29528): [onServicesDiscovered] count: 4 status: 0
D/BluetoothGatt(29528): setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
E/flutter (29528): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)
E/flutter (29528): #0      StandardMethodCodec.decodeEnvelope
package:flutter/…/services/message_codecs.dart:653
E/flutter (29528): #1      MethodChannel._invokeMethod
package:flutter/…/services/platform_channel.dart:296
E/flutter (29528): <asynchronous suspension>
E/flutter (29528): #2      BluetoothCharacteristic.write
package:flutter_blue_plus/src/bluetooth_characteristic.dart:117
E/flutter (29528): <asynchronous suspension>
E/flutter (29528): #3      BlueService.write
package:wisdoma_app/utils/blue_service.dart:116
E/flutter (29528): <asynchronous suspension>
E/flutter (29528):
D/FlutterBluePlugin(29528): [onDescriptorWrite] uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/BluetoothAdapter(29528): isLeEnabled(): ON

liyeatgithub, 2023-01-29:
same problem,when i connected the host device.And i  send data is success and get the response.But when i send data again this problem hanpend.

Lokendra16, 2023-03-09:
I am also getting write characteristics error.

cvanbeek13, 2023-04-10:
I'm also getting this same error.

felipeolliveira, 2023-05-24:
I'm also getting this same error and adding delay before it is not enough to "solve".
I added before and after `characteristic.write()` and it worked for me.
It seems to me that the *Future* in this function is not working as it should...

vrajshah24, 2023-05-24:
> [felipeolliveira](/felipeolliveira)

Can you pls share me that code,that would be much helpful, i have tried but seems like its not working for me so for that reason can you pls share me the code

nangbanmai1907, 2023-05-29:
I need add sleep when write and read
Ex:
```
Future _writeBLE(
      BluetoothCharacteristic characteristic, List<int> hexByte) async {
    try {
      sleep(const Duration(seconds: 1));
      await characteristic.write(hexByte, withoutResponse: true);
    } catch (e) {
      print('-------------------error _writeBLE: $e');
    }
  }

  Future<String> _readBLE(BluetoothCharacteristic characteristic) async {
    try {
      sleep(const Duration(seconds: 1));
      List<int> dataByte = await characteristic.read();
      String dataString = String.fromCharCodes(dataByte);
      return dataString;
    } catch (e) {
      print('-------------------error _readBLE: $e');
    }
  }
```

felipeolliveira, 2023-06-02:
> > [felipeolliveira](/felipeolliveira)
> 
> Can you pls share me that code,that would be much helpful, i have tried but seems like its not working for me so for that reason can you pls share me the code

I've been changed my code and also works with one delay, but this delay must be more than 500 milliseconds to work:  
```dart
Future<void> send(List<int> bytes) async {
    await Future.delayed(const Duration(seconds: 1));
    _writeCharacteristic?.write(bytes, withoutResponse: true);
  }
```

I don't like this solution with sleeps, but is working until now.

chipweinberger, 2023-07-20:
same issue here: https://github.com/boskokg/flutter_blue_plus/issues/385

please try 1.7.6+, feel free to reopen if you still have problems.

closed 2023-07-20 by chipweinberger