chipweinberger/flutter_blue_plus #133: Listener remove

massimilianochiodi, 2022-09-08:
is possible remove listener for an characteristic ? 
Plese...

navaronbracke, 2022-09-17:
@massimilianochiodi Assuming you use the following to listen to changes for a characteristic:
```
await characteristic.setNotifyValue(true);
characteristic.value.listen((value) {
    // do something with new value
});
```

You should instead cancel the `StreamSubscription` that is returned by `listen((value){...})`

I.e.

```
import 'dart:async';

StreamSubscription<List<int>>? subscription;

await characteristic.setNotifyValue(true);
subscription = characteristic.value.listen((value) {
    // do something with new value
});

subscription?.cancel();
```

AlexQiulin, 2023-01-08:
Very good solution

AlexQiulin, 2023-01-08:
> @massimilianochiodi Assuming you use the following to listen to changes for a characteristic:
> 
> ```
> await characteristic.setNotifyValue(true);
> characteristic.value.listen((value) {
>     // do something with new value
> });
> ```
> 
> You should instead cancel the `StreamSubscription` that is returned by `listen((value){...})`
> 
> I.e.
> 
> ```
> import 'dart:async';
> 
> StreamSubscription<List<int>>? subscription;
> 
> await characteristic.setNotifyValue(true);
> subscription = characteristic.value.listen((value) {
>     // do something with new value
> });
> 
> subscription?.cancel();
> await characteristic.setNotifyValue(false); // Not sure if this is needed
> ```

good

closed 2023-01-08 by boskokg