chipweinberger/flutter_blue_plus #147: I found three serious bugs with this Bluetooth plug-in. I hope the author can solve them when updating it next time

AlexQiulin, 2022-10-10:
1：
The first problem is that errors are often reported when finding eigenvalues. Although errors are not always reported, the probability of error is very high
[image]

2：
The second problem is that when Bluetooth is disconnected and reconnected for many times, the return of Bluetooth notify will be infinitely overlapped. Unless you exit the APP and kill the process, you can restart the APP again
[image]

3：
The third problem is that when the Bluetooth device is normally connected to the APP, manually go to the mobile phone system and turn off Bluetooth. This will cause the Bluetooth device to be disconnected all the time, and cannot be connected even after recompilation, which is very strange.
[image]

If you have corresponding solutions, please advise. Similarly, if you have the same problems, I hope the author can fix these bugs

This is the sample code I wrote：
https://github.com/AlexQiulin/flutter-blue-plus/blob/main/main.dart

WeiKuang, 2022-10-11:
bad code style

ifredom, 2022-10-20:
点击开始扫描 startScan()，直接就卡死了  ，
[image]

sharathltts, 2022-11-02:
same issue

971586331, 2022-12-09:
The same problem, there were these problems in the previous version 0.8, the strange thing is that it has not been changed, I wonder if there is a problem with the way I use it

liyao20190808, 2023-01-17:
same issue

chipweinberger, 2023-07-20:
please try 1.7.6+, feel free to reopen if you still have problems.

closed 2023-07-20 by chipweinberger