chipweinberger/flutter_blue_plus #148: Notes on Android12 release APK (Build APK)

AlexQiulin, 2022-10-11:
For Android 12, the APK file after the official package (Build APK) release cannot be searched for Bluetooth devices and the system Bluetooth solution cannot be opened
[image]

forte-sama, 2022-10-12:
Something related happens to me, a debug build behaves like a charm, in release builds the app fails to discover services on the peripheral device.

Calvin-Ho, 2022-10-17:
I have just fixed this in my own project.

You can still use `shrinkResources` and `minifyEnabled` as shown here:
```
buildTypes {
        release {
            signingConfig signingConfigs.release

            minifyEnabled true

            shrinkResources true

            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
```
but you need to make sure to update your Proguard rules in `android/proguard-rules.pro`:

```
...

# flutter_blue_plus rules
-keep class com.boskokg.flutter_blue_plus.** { *; }
-keepclassmembernames class com.boskokg.flutter_blue_plus.* { *; }
````

This will prevent the classes and relevant members from being minified which means that they can be invoked as normal.

kashjack, 2022-11-03:
It's working in my case, thank you

ChampiMagic, 2022-11-05:
I dont have the "proguard-rules.pro" file on me android folder, were can I found it?

axelvugts, 2022-11-16:
> I dont have the "proguard-rules.pro" file on me android folder, were can I found it?

You have to manually create the file if it doesn't exist yet.

closed 2022-12-01 by boskokg

svprdga, 2023-01-13:
> buildTypes {
>         release {
>             signingConfig signingConfigs.release
>
>             minifyEnabled true
>
>             shrinkResources true
>
>             proguardFiles [...]

Thank you so much for this information.

Without a doubt, this information should appear prominently in the README.md for this plugin.

zhaojun1217, 2023-03-20:
good issue  , thank every one
i used
-keep class com.boskokg.flutter_blue_plus.** { ; }
-keepclassmembernames class com.boskokg.flutter_blue_plus. { *; }