chipweinberger/flutter_blue_plus #149: BluetoothState returns always null

JP3871, 2022-10-16:
Hi,

Code below returns always null for FlutterBluePlus.instance.state even if I activate Bluetooth

Xiaomi Mi Note 10 lite
Android : version 11
Flutter version 3.3.4
Flutter blue plus version 1.3.1

  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      color: Colors.lightBlue,
      home: StreamBuilder<BluetoothState>(
          stream: FlutterBluePlus.instance.state,
          initialData: BluetoothState.unknown,
          builder: (c, snapshot) {
            final state = snapshot.data;
            if (state == BluetoothState.on) {
              return const FindDevicesScreen();
            }
            return BluetoothOffScreen(state: state);
          }),
    );
  }

hoffmannkrzysztof, 2022-10-20:
Having the same issue on Redmi 8A android 10, miui 12.5.3. On other devices eg. Samsung Galaxy Tab S7 FE works fine.

JP3871, 2022-10-21:
@hoffmannkrzysztof
> I figured out what's was wrong. In the settings -> location. There is addition permission to enable scan wifi and bluetooth devices by apps. Default it is off, I guess

You're the best! You're right, The bluetooth option is disable by default
Thank you very much

closed 2022-10-21 by JP3871