chipweinberger/flutter_blue_plus #150: Calling disconnect() doesn't really disconnect, but it does close() BluetoothGatt which prevents future reconnection, catch-22

zyphlar, 2022-10-20:
I'm using the nRF Connect app to debug why a first scan/connect/read works on Android (and, generally, also on iOS) but subsequent attempts fail. It appears that the bluetooth LE device remains connected indefinitely even after the device.disconnect(); call, yet BluetoothGatt close() and unregisterApp() are called which results in the below error on next scan/connect attempt. Overall, disconnect seems extremely buggy and error-prone. (A second attempt at scanning/connecting after this error seems to work, however it works best when the device has been force-disconnected by external means, which is very cumbersome.)

Using flutter_blue_plus version 1.3.1

```
PlatformException PlatformException(discover_services_error, no instance of BluetoothGatt, have you connected first?, java.lang.Exception: no instance of BluetoothGatt, have you connected first?
I/flutter ( 9439): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.locateGatt(FlutterBluePlusPlugin.java:734)
I/flutter ( 9439): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:406)
I/flutter ( 9439): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:262)
I/flutter ( 9439): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
I/flutter ( 9439): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:319)
I/flutter ( 9439): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
I/flutter ( 9439): 	at android.os.Handler.handleCallback(Handler.java:938)
I/flutter ( 9439): 	at android.os.Handler.dispatchMessage(Handler.java:99)
I/flutter ( 9439): 	at android.os.Looper.loopOnce(Looper.java:226)
I/flutter ( 9439): 	at an
```

zyphlar, 2022-10-25:
Might be related to #48

damowangcaixiaofeng, 2022-10-28:
Same problem, why, go to send message page error

closed 2022-12-01 by boskokg