chipweinberger/flutter_blue_plus #151: Samsung Galaxy S10e not scanning

GiovanniRaseraF, 2022-10-21:
Hi, i built the example app for release and tested it on 4 different devices, but when testing on Samsung Galaxy S10e and S9 the scan does not work.

Can it be the version of the BLE ?

kekko7072, 2022-10-22:
+ 1

zyphlar, 2022-10-24:
I also can't get the example app in the latest commit to produce any scan results, even though I've used flutter_blue_plus releases in my own apps successfully. Samsung Galaxy S21 stock/latest

GiovanniRaseraF, 2022-10-24:
Maybe the solution can be [Issue 152](https://github.com/boskokg/flutter_blue_plus/issues/152), going to try in the next days. 
I tried the app using IPhone 13 and is has no problem connecting to BLE.

kekko7072, 2022-10-24:
@GiovanniRaseraF thanks, fo me also iOS is working fine but android not. Can you explain a little better this is my code to scan, how should I change the start scan?

```
class BLEFindDevices extends StatelessWidget {
  const BLEFindDevices({Key? key}) : super(key: key);

  @override
  Widget build(BuildContext context) {
    return Scaffold(
      body: RefreshIndicator(
        onRefresh: () => FlutterBluePlus.instance
            .startScan(timeout: const Duration(seconds: 4)),
        child: SingleChildScrollView(
          child: Column(
            children: <Widget>[
              StreamBuilder<List<BluetoothDevice>>(
                stream: Stream.periodic(const Duration(seconds: 2))
                    .asyncMap((_) => FlutterBluePlus.instance.connectedDevices),
                initialData: const [],
                builder: (c, snapshot) => Column(
                  children: snapshot.data!
                      .map((d) => ListTile(
                            title: Text(d.name),
                            subtitle: Text(d.id.toString()),
                            trailing: StreamBuilder<BluetoothDeviceState>(
                              stream: d.state,
                              initialData: BluetoothDeviceState.disconnected,
                              builder: (c, snapshot) {
                                if (snapshot.data ==
                                    BluetoothDeviceState.connected) {
                                  return ElevatedButton(
                                    child: const Text('OPEN'),
                                    onPressed: () => Navigator.of(context).push(
                                        MaterialPageRoute(
                                            builder: (context) =>
                                                BLEDeviceScreen(device: d))),
                                  );
                                }
                                return Text(snapshot.data.toString());
                              },
                            ),
                          ))
                      .toList(),
                ),
              ),
              StreamBuilder<List<ScanResult>>(
                stream: FlutterBluePlus.instance.scanResults,
                initialData: const [],
                builder: (c, snapshot) => Column(
                  children: snapshot.data!
                      .where((element) =>
                          element.device.name == kDeviceModelTKR1A1)
                      .map(
                        (r) => ScanResultTile(
                          result: r,
                          onTap: () async => await r.device.connect().then(
                              (value) => Navigator.of(context).push(
                                      MaterialPageRoute(builder: (context) {
                                    return BLEDeviceScreen(device: r.device);
                                  }))),
                        ),
                      )
                      .toList(),
                ),
              ),
            ],
          ),
        ),
      ),
      floatingActionButton: StreamBuilder<bool>(
        stream: FlutterBluePlus.instance.isScanning,
        initialData: false,
        builder: (c, snapshot) {
          if (snapshot.data!) {
            return FloatingActionButton(
              onPressed: () => FlutterBluePlus.instance.stopScan(),
              backgroundColor: Colors.red,
              child: const Icon(Icons.stop),
            );
          } else {
            return FloatingActionButton(
                child: const Icon(Icons.search),
                onPressed: () => FlutterBluePlus.instance
                    .startScan(timeout: const Duration(seconds: 4)));
          }
        },
      ),
    );
  }
}

```

GiovanniRaseraF, 2022-10-24:
@kekko7072 i tried your code without the filter:
`.where((element) =>
                          element.device.name == kDeviceModelTKR1A1`
and it works only using Huawwi p20 lite and IPhones 

i did to understand what you want to change in .startScan,
please contact me at giovannirasera3@gmail.com (sono italiano) :)

kekko7072, 2022-10-24:
Hey thank's @GiovanniRaseraF for your help I discover my issue was on settings the name on the BLE Device. Thanks a lot for support.

closed 2022-12-01 by boskokg