chipweinberger/flutter_blue_plus #154: BluetoothGatt.writeCharacteristic is depreacated in Android API 33

choim4389, 2022-10-25:
It seems to be deprecated in latest android api.  ([refer](https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#writeCharacteristic(android.bluetooth.BluetoothGattCharacteristic)))

Is it any plan to cover that issue?

sakina1403, 2023-02-16:
Thanks for helping me understand why it's failing on Android 13. Suprisingly its deprecated in October and still no update on the plugin :( I am trying to resolve it now, will update.

closed 2023-05-03 by chipweinberger

nilsorathiya, 2024-05-02:
@sakinaboriwala 
I am having the same issue. can you help me with how I can figure it out?

I was using the Blufi plugin but it is asking for Bluetooth permission every time I open the app and again the main issue is, an outdated version of Blufi.

It will be really helpful if you can help it out!