chipweinberger/flutter_blue_plus #159: it support multiple devices？

damowangcaixiaofeng, 2022-11-04:


coolkiecat, 2022-11-11:
yes at least in android, see the android code, use FlutterBlue.instance.connected will return java List ((BluetoothManager) application.getSystemService(Context.BLUETOOTH)).getConnectedDevices(...),
then when you connect a device, just put it in the specified Map<macid, device> cache in your dart code to manage mutilple device.

MrCsabaToth, 2022-11-19:
I can confirm that my application sometimes may connect to three different devices at once (speed sensor, cadence sensor, and HRM) all streaming data simultaneously. Not outstandingly huge data bandwidth though in my case, I haven't ever dealt with devices like listening to music.

coolkiecat, 2022-11-19:
[自动回复]您好，我已收到你的邮件，这是邮箱自动回复

coolkiecat, 2022-11-19:
> I can confirm that my application sometimes may connect to three different devices at once (speed sensor, cadence sensor, and HRM) all streaming data simultaneously. Not outstandingly huge data bandwidth though in my case, I haven't ever dealt with devices like listening to music.

In my cases, it will cut the package and I need to regenized whether it is a single one or a fragmented one, i think it dependts on "MTU", which means "max transport unit" i guess.
So the important thing is,  while you are listening the characteristic, just check the package, if it isnt  a single one, put it in the stack like "map<macid, stack>" for mutiple devices, when the tail fragment or single is being listented,  notify all listener in the map<macid, List<void Function(package)>> which registe before listen.

MrCsabaToth, 2022-11-20:
> In my cases, it will cut the package and I need to regenized whether it is a single one or a fragmented one, i think it dependts on "MTU", which means "max transport unit" i guess. So the important [...]

Sounds like this is a connection to a specific device. How multiple devices come into the picture? Is the title of the issue might be misnomer (so instead of multiple devices you are talking about multiple packet fragments)?
Also what you describe (macid stack) is in the native (Android / iOS) layer of the plugin or at the Flutter layer of the plugin?

coolkiecat, 2022-11-20:
> Sounds like this is a connection to a specific device. How multiple devices come into the picture? Is the title of the issue might be misnomer (so instead of multiple devices you are talking about [...]

The macid stack is not defined in any layer, but you need to declare a global one in your bluetooth module so that you can handle it after some packages which i call these head, middle1, middle2, ...tail fragment that should needs to be merge, and caculate its crc16/crc32 to make sure it would not lose package.
The title "support multiple devices?" not perfectly right. "Android/ios support ...?" or "support multiple connection manage" is more accurate, the first is yes, but the second is no, you have to write a multiple device manager that contains QOS, Event/Listener/Dispatcher, unicode/ascii handler, Cache... by yourselfes, its complicated

closed 2022-12-01 by boskokg

MrCsabaToth, 2022-12-01:
What you describe sounds like to me out of jurisdiction of a top layer bluetooth plugin and is the responsibility of a lower level layer (within the native Bluetooth stack of Android OS or iOS respectively) of the tech stack. I'm thinking in ISO-OSI networking layers here. When the data packet arrives at the Dart section of the plugin (and then your app) it should be already "processed". MTU and similar stuff are responsibility of lower level transport layers and not the plugin. This plugin is essentially a shim to native calls. Essentially it defines protobuf objects and translates the communication to a common language between the iOS and Android, however it mostly deals with handling the callbacks and calls to the native stack layer.

Maybe if you describe your need more precisely, maybe some code examples or something what you deal with. Is your application some proprietary BT hardware, some custom Arduino stuff, or some consumer electric connection?

coolkiecat, 2022-12-02:
> @coolkiecat 
> What you describe sounds like to me out of jurisdiction of a top layer bluetooth plugin and is the responsibility of a lower level layer (within the native Bluetooth stack of Android OS or iOS respectively) of the tech stack. I'm thinking in ISO-OSI networking layers here. When the data packet arrives at the Dart section of the plugin (and then your app) it should be already "processed". MTU and similar stuff are responsibility of lower level transport layers and not the plugin. This plugin is essentially a shim to native calls. Essentially it defines protobuf objects and translates the communication to a common language between the iOS and Android, however it mostly deals with handling the callbacks and calls to the native stack layer.
> 
> Maybe if you describe your need more precisely, maybe some code examples or something what you deal with. Is your application some proprietary BT hardware, some custom Arduino stuff, or some consumer electric connection?

I did test multiple device in android with java version, an android only have one BluetoothAdaptor object, But it could has many BluetoothGate object that help to connect many BLE devices. Its one adaptor, many gatt; one gatt, one connection. This issue title shoud be "how can android/ios connect multiple devices? is someone could teach me how to do that?"If the questioner read flutter_blue android/ios part code , or test it in native part, he/she will not ask this questions. In java version of flutter_blue plugin, it did put a BluetoothGatt in a deviceCache map when called device.connect(), and manage it after.This means it supports multiple devices.