chipweinberger/flutter_blue_plus #160: Possible to save list of connected devices, restart app, and then reconnect using list of saved devices?

alexvnesta, 2022-11-05:
Hi, 

I have done some pretty extensive searching and I can not figure out an IOS and Android compatible way to save my currently connected devices, close the app completely, and then re-connect to those saved devices.

I have tried writing my list of BluetoothDevice objects to a file, but the only practical way to do this in dart/flutter seems to be encoding the data in JSON format. To do this, I would need to save the connected device's mac address as a string, and then make a new BluetoothDevice object, assign the name, id, and type, and re-connect to it. I am unable to do this though, because I can not create an unnamed constructor for BluetoothDevice.

`      BluetoothDevice myDevice = BluetoothDevice(
        id: DeviceIdentifier("MAC address"),
        type: BluetoothDeviceType.le,
        name: "Device Name",
      );
      await myDevice.connect();
`

`
The class 'BluetoothDevice' doesn't have an unnamed constructor. (Documentation)  Try using one of the named constructors defined in 'BluetoothDevice'.
`

Please, if there is another workaround/solution, I would really appreciate the help.

Thanks,
Alex

LeoBound, 2022-11-05:
To achieve this I save the MAC to a shared preferences value on connection, then on next app startup I read the MAC, start a scan, and connect to the device with the matching address if I find it.

This works on Android, and I think iOS but I haven't tested it in a while.

alexvnesta, 2022-11-05:
Thank you for the suggestion @LeoBound .

would you mind sharing a quick code snippet to create a Bluetooth device and connect to it using just the MAC address? Or do you have a file with this code that you can share? 

I have been trying to do this, but I can’t figure it out.

Thanks,
Alex

alexvnesta, 2022-11-06:
UPDATE: I got this working on both iOS and android. I did the following:

add to imports:
```
import 'package:flutter_blue_plus/gen/flutterblueplus.pb.dart' as ProtoBluetoothDevice;
```

Make a list to store your currently connected devices, and specifically save the MAC address (Android)/ uuid (ios) - the same code works for both, but you will notice the IDs are different in the debug output text.

```
  List<BluetoothDevice>? connectedDevicesList;
  List<String> savedDeviceMacAddresses = [];

  void saveConnectedDevices(List<BluetoothDevice> deviceList){
    for (BluetoothDevice device in deviceList){
      print(device.id.id.toString());
      savedDeviceMacAddresses.add(device.id.id);
    }
    for (String deviceID in savedDeviceMacAddresses){
      print("Successfully saved $deviceID");
    }
  }

  void reconnectSavedDevices(List<String> savedDeviceIDs) async{
    for (var deviceID in savedDeviceIDs) {
      print("going to connect to device $deviceID");
      var protoBt = ProtoBluetoothDevice.BluetoothDevice(
          remoteId: deviceID
      );
      late var d = BluetoothDevice.fromProto(protoBt);
      await d.connect();
    }
  }
```

please note, I got the example code for making a prototype BT device from here:
https://github.com/pauldemarco/flutter_blue/issues/333

Maybe this will help someone in the future.

Please note, I did not actually save the devices to local storage yet, but it is relatively simple to save a list of strings to local storage with flutter. Just google it.

closed 2022-11-06 by alexvnesta

boskokg, 2022-11-06:
Just to note that this scenario maybe will not work if android cache is cleared (eg turning off bluetooth and turning on again), because you call connect with property auto connect false, that don't call implicit scan and expects that bluetooth cache has that device, and probably device will not be in the cache because scan is not called.

WeiKuang, 2022-11-17:
ios is not work.