chipweinberger/flutter_blue_plus #162: PlatformException(locateCharacteristic, service could not be located on the device, null, null)

kamilszpila, 2022-11-10:
It seems that we receive this error while trying to read or write to characteristic just after disconnection.
There is some solution suggested here https://github.com/trueb2/flutter_blue/pull/5

MrCsabaToth, 2022-11-19:
Interesting, the referred PR is a fork of the boskosg/flutter_blue and not the boskosg/flutter_blue_plus. I wonder if that was merged there, why is it not in this (flutter_blue_plus) project, and what other PRs merged into that project are not in the plus project.

kamilszpila, 2023-01-18:
@boskokg @MrCsabaToth can you please fix it in this repository as well?

MrCsabaToth, 2023-01-18:
> @boskokg @MrCsabaToth can you please fix it in this repository as well?

I don't have time for this now, can you craft a PR if you are at it?

closed 2023-03-16 by boskokg