chipweinberger/flutter_blue_plus #166: Android 12: Failed to handle method call / NullPointerException

axelvugts, 2022-11-18:
Hi!

I'm currently working on an app that scans for BLE devices in the background using `flutter_blue_plus`.
Last night, my development phone updated from Android 11 to Android 12 and now my app throws an error in the background.

`E/MethodChannel#flutter_blue_plus/methods(15115): Failed to handle method call
E/MethodChannel#flutter_blue_plus/methods(15115): java.lang.NullPointerException: Attempt to invoke interface method 'android.app.Activity io.flutter.embedding.engine.plugins.activity.ActivityPluginBinding.getActivity()' on a null object reference
E/MethodChannel#flutter_blue_plus/methods(15115): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.ensurePermissionBeforeAction(FlutterBluePlusPlugin.java:722)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$1$com-boskokg-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:260)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda3.op(Unknown Source:6)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.ensurePermissionBeforeAction(FlutterBluePlusPlugin.java:727)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:258)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:262)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:319)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at android.os.Handler.handleCallback(Handler.java:938)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at android.os.Looper.loopOnce(Looper.java:201)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at android.os.Looper.loop(Looper.java:288)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at android.app.ActivityThread.main(ActivityThread.java:7874)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at java.lang.reflect.Method.invoke(Native Method)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
E/MethodChannel#flutter_blue_plus/methods(15115): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
I/flutter (15115): Error starting scan.`

Any suggestions on what could be the problem?

MrCsabaToth, 2022-11-19:
How do you scan in the background? Background work on Android is a native feature and I'm curious how you approach that.

axelvugts, 2022-11-20:
> How do you scan in the background? Background work on Android is a native feature and I'm curious how you approach that.

I use the [flutter_foreground_task](https://github.com/Dev-hwang/flutter_foreground_task) plugin for that.

MrCsabaToth, 2022-11-20:
I always wanted to try that (I've looked at multiple plugins). I would not need true boot time background capabilities but only the capability that my app would keep recording the Bluetooth data even if it's not in the foreground. So this worked fine for you before Android 12?

axelvugts, 2022-11-20:
> I always wanted to try that (I've looked at multiple plugins). I would not need true boot time background capabilities but only the capability that my app would keep recording the Bluetooth data even if it's not in the foreground. So this worked fine for you before Android 12?

Yes it definitely worked well, but It did take me some time to figure out hat the "Location Always" permission is required for it to work.
You should also know that it only works on Android (on iOS the background task will be killed by the system after a few seconds) and that I haven't tried to use it in production yet, so your mileage may vary :)

MrCsabaToth, 2022-11-20:
> Yes it definitely worked well, but It did take me some time to figure out hat the "Location Always" permission is required for it to work. You should also know that it only works on Android (on iOS [...]

Thanks for the info. Battery optimization is an unregulated area on Android as well, see https://dontkillmyapp.com/

axelvugts, 2022-11-21:
After some further investigation, it turns out the culprit was the `BLUETOOTH_CONNECT` permission that is required on Android >= 12 even if you only scan for devices. Adding that to the Android manifest solved the problem for me.

closed 2022-11-21 by axelvugts

watsmyname, 2022-12-19:
> Hi!
>
> I'm currently working on an app that scans for BLE devices in the background using `flutter_blue_plus`. Last night, my development phone updated from Android 11 to Android 12 and now my app [...]

Can you please share me your code, how you are using flutter_blue_plus with flutter_foreground_task?