chipweinberger/flutter_blue_plus #167: Is it possible to scan in background?

baranacikgoz, 2022-11-19:
I am scanning devices to get their rssi values. App is successfully working while in foreground, but when i try to run with some background related pub.dev packages, scan is not working in background

axelvugts, 2022-11-20:
You should ensure that you have the right permissions. You need to get the user to disable battery optimisation for your app and you'll need location permissions for in the background.

closed 2022-12-01 by boskokg

jawadabbasi923, 2023-02-27:
@axelvugts can you share any solution for background scanning and connection of BLE device.
My app is working fine in the foreground but I want to get the connected device data in the background or app is minimised.
Thank you.